uavpal/disco4g #7: Possible to Install on Bebop 2?

teja235, 2018-05-09:
I’m reading through the code and at first glance, it doesn’t appear that there’s any reason this mod wouldn’t work on Parrots Bebop 2 as well. Do you know if any porting would be required?

Telepilote, 2018-05-09:
I had wondered the same, although the long distance benefits are less this mod would reduce drop outs as the operator will not need to maintain line of sight and will add tracking for recovery. Both very handy improvements.

teja235, 2018-05-09:
The WiFi signal from Skycontroller to Bebop is very unreliable and requires direct line of site between controller and drone. LTE maybe a better option than WiFi in a location with a large number of obstacles blocking the signal, or in the case of trying to fly the drone beyond line of sight.

teja235, 2018-05-09:
Any information about the technical aspects? I can modify the code But this is my first time coding on parrots platform. What’s the difference between disco and bebop firmware?

mainframe, 2018-05-10:
We dont have Bebop2 to test/verify - but as long its CHUCK and similar firmware it should be possible - and maybe even without any porting.

Telepilote, 2018-05-10:
It’s not CHUCK so there will be differences and similarities.

teja235, 2018-05-10:
@mainframe see that’s what I thought. Will try with bebop 2 right after I get it working on my disco - right after finals week. 😓

mainframe, 2018-05-10:
Bebop 2 hardware is ARM based? Does it have USB port?

mainframe, 2018-05-10:
This spec seems similar to CHUCK - ie its still P7 SoC and Linux OS on ARM.

teja235, 2018-05-11:
It is the same SoC, among many other shared components, so I’m positive it will work but maybe not “as is”. I’m waiting on my modem to arrive - gonna get to work figuring this out soon!

Telepilote, 2018-05-11:
Bebop 2 has a single micro USB port accessible at the rear for updates, image transfer, increase storage capacity and even operate a thermal camera. https://www.parrot.com/business-solutions-us/parrot-professional/parrot-bebop-pro-thermal

KillahDT, 2018-05-11:
Will this brick my Bebop 2 if i try to install this mod?
Very nice work ice searched for it.
Ive also a Bebop 1 but i think the system is more different to Bebop 2 and C.H.U.C.K. one the Disco.

greetz

teja235, 2018-05-11:
I don’t think it would be a permanent brick... worst case it doesn’t work and you’d have to reinstall the firmware. But my plan was to go through the code line by line and see if the files that are being changed are in the same spots if that makes sense lol. I wouldn’t recommend just straight up installin it to see what happens only Bc it’s a $300 drone, but if you’re going to do it then please let me know what happens.

KillahDT, 2018-05-12:
i can test it but how to unbrick if it dont work

teja235, 2018-05-12:
It kinda depends on what happens lol. You could just do it and if it flops
we’ll figure out how to fix it then 🤷🏽‍♂️

uavpal, 2018-05-12:
You can try it on Bebop2, I don't see any reason why it would not work. There might be minor changes required to the scripts (apart from replacing text outputs "disco" with "bebop"), but it should certainly not brick your drone! I could imagine the custom kernel modules I cross-compiled will not work on the Bebop2. Might require re-compilation. Worst case: it will just not work, I don't foresee a "bricking" scenario, which would render the drone completely useless.
However, do not try the Disco4G scripts on Bebop1 -OR- on Bebop2 with the original Skycontroller (which is also still sold as a bundle). Both will not work for sure (without bigger changes).

There are a few more people asking for it in the Slack workspace. I do not have a Bebop2 myself, but I believe it would be very useful! Unlike the disco, failover scenarios for a quad are not that critical (as they just hover whenever a connection loss occurs).

Please drop us a note if you get it to work. Happy to add it to the instructions and accept pull requests if code changes are required.

Thanks, SoftIce

KillahDT, 2018-05-12:
I will install it later i have the skycontroller 2 
But i havent a huawei stick to test 
i will buy one ending this month
if the drone still works after install i will update to this thread

uavpal, 2018-05-12:
great!! 👍 
(please also update if it does not) 😃

KillahDT, 2018-05-12:
I need help.
i cannot copy the disco4g-master folder to my bebop
Most of the files are copied but many arent.
i get errors on filezilla and winscp on file
usbserial.ko
usb_wwan.ko
xt_tcpudk.ko
x_tables.ko
and more.

greetz

uavpal, 2018-05-13:
what error? is there enough disk space?

KillahDT, 2018-05-13:
Yeah the drone has 7 GB free space.
It shows me that file already exist and cannot overwrite the file.
if i delete the disco 4g folder and ty it again its the same problem.
error 553 but im not sure

uavpal, 2018-05-13:
can you telnet into your bebop and post the output of `df -h` ? thanks

KillahDT, 2018-05-13:
BusyBox v1.25.0 (2018-02-14 09:37:35 UTC) built-in shell (ash)
Enter 'help' for a list of built-in commands.

/ # df -h
Filesystem                Size      Used Available Use% Mounted on
ubi1:system              42.2M     38.3M      1.7M  96% /
udev                    144.9M    792.0K    144.2M   1% /dev
shm                     145.0M     36.0K    145.0M   0% /dev/shm
tmp                     145.0M     32.0K    145.0M   0% /tmp
ubi0:factory              4.8M    108.0K      4.4M   2% /factory
ubi2:data                 9.0M      8.9M         0 100% /data
ubi2:update              28.0M     32.0K     26.5M   0% /update
/dev/mmcblk0              7.2G    148.4M      7.0G   2% /data/ftp/internal_000
/ #

uavpal, 2018-05-13:
Yep, as suspected:
/data and therefore /data/ftp has **0MB** free (of total 9MB)
/data/ftp/internal_000 has 148MB free (of total 7.2GB) - it's a different mountpoint
Seems the Disco has more space available in /data/ftp than Bebop2.
The softmod requires 12MB.

I suggest you use FileZilla to delete /disco4g-master.
You can try to use /internal_000 instead. But this will require changes to the manual instructions as well as installer script!

KillahDT, 2018-05-13:
Thats to hard for me to do 
i havent the knowledge for it

uavpal, 2018-05-13:
I haven't the bebop2 for it :(

KillahDT, 2018-05-14:
Should i send my bebop 2 to you?
😅

teja235, 2018-05-14:
I gottchu @KillahDT. @UAVPal thanks for staying active on this, I’m sure we can figure it out together. I’ll make those changes to the script and give it a run, modems coming in 2 days.

KillahDT, 2018-05-14:
Thats cool im happy to hear that.
Waiting for your test 😄

ToddTsao, 2018-05-17:
1st try run fail. Just share my fail experience.
IP set is the same as Disco.
FTP transfer file to BB2: Done
Enable BB2 telnet mode: press 4 times power button after power on.
1st time install process: BB2 space is lock: Fail
-------------------------------------
Creating ppp directory
Creating symlink for ppp-lte settings
Creating symlink udev rule
Creating symlink for ntpd's config file
Remounting filesystem as read-only
Creating zerotier-one directory
mkdir: can't create directory '/data/lib/zerotier-one': No space left on device
Creating symlink for zerotier-one's local config file
ln: /data/lib/zerotier-one/local.conf: No such file or directory
Removing uavpal softmod installation files
------------------------------------------------------------
Add some command to unlock the space.
   1. cd /etc/colibry/common/ 
   2. mount -o remount,rw /
Install process is done after unlock
Reboot BB2
The 4G dongle LED is no light. No driver. 
How to enable usb port?  what's happen?
I don't know.

KillahDT, 2018-05-18:
Have you a sim card inside the modem?
I doesnt have a 4g dongle yet but maybe its installed correctly on your drone and only the sim have to sit in the dongle

ToddTsao, 2018-05-18:
I have install the SIM card. I use the same SIM card for my DISCO-4G & BB2. The Disco-4G is no problem.
The 4G dongle is no power on for BB2.

KillahDT, 2018-05-19:
Maybe the Bebop 2 only needs a simple software modification to the disco4g master mod.

ToddTsao, 2018-05-21:
KillahDT: I have find a solution to enable the BB2 USB port. 
FYI
https://community.parrot.com/t5/Bebop-Drone/Bebop-USB-port-what-type-is-that-logically-SOLVED/m-p/133214

KillahDT, 2018-05-22:
Will anybody update the mod for the bebop2?

job2310, 2018-05-22:
FYI, result of uname -a

Linux Bebop2-104893 3.4.11+ #1 SMP PREEMPT Wed Feb 14 09:44:38 UTC 2018 armv7l GNU/Linux

ToddTsao, 2018-05-28:
Currently, we finish below:
Copy Disco-4G to /internal000/
Change file path
Install it to BB2
Enable micro USB port (4G modem power on)

Issue:
BB2 still cannot to drive the 4G modem. It's no connect to Zerotier from BB2.

Need help
Any suggest or command for that.

job2310, 2018-05-28:
@ToddTsao the mod will not work on the BB2 without several kernel modules being recompiled, specifically the modules used for ppp which drive the 4G connection to the mobile network.

KillahDT, 2018-05-28:
Is it much of work to do this?

job2310, 2018-05-28:
I believe it takes a bit of time to do.  The main issue is knowing how to compile the modules for the correct kernel version.  That's outside my skillset.  @uavpal has kindly agreed to do it for us as and when he has time.  It may take several attempts to get the modules compiled and working on the BB2.  I'll post back here if/when we make progress.

KillahDT, 2018-05-29:
Im happy to hear that and will wait for your update.

greetz

uavpal, 2018-06-04:
[image]

Teaser...

AlaskaJason1090, 2018-06-04:
I have waited for this moment for so long. Thank you all who helped build this. I cant wait to try this here in 
Alaska. We have tons great places to fly and this will make it so much better.

KillahDT, 2018-06-04:
🔥🙂🔥

Telepilote, 2018-06-04:
Given the great support I would like to further encourage users to hit the donate icon on the code page, I’ve also copied the link here: 

https://www.paypal.com/donate/?token=71em4QrgOUbXVs0Smqd9m6Qy6ZQWWP8jB2Am6bwlydmkhbJmFx0jKKOqgdOWtrMtZFID40&country.x=US&locale.x=US

KillahDT, 2018-06-05:
Donating link doesnt work for me.

uavpal, 2018-06-06:
You can try it from here: https://github.com/uavpal/disco4g#is-it-really-free-are-you-crazy
If not, what error do you get?

OpenUAS, 2018-06-06:
Well if you want more ;) https://www.youtube.com/watch?v=qVogmRistcw

job2310, 2018-06-06:
FYI, Bebop2 ... the method of killing /usr/bin/usb_mode and switching usb to host doesn't work with the latest firmware as /usr/bin/usb_mode no longer exists.  Looks like Parrot are onto us. :)

KillahDT, 2018-06-07:
Which firmware is supported?
I will look which i have installed after work.

job2310, 2018-06-07:
[image]

Live streaming via 4G connection to Bebop2.

I've worked out a process where a Raspberry Pi Zero W can be used to connect to the Bebop2 hotspot and share it's internet connection with the Bebop2.  ZeroTier is installed on the Bebop2 to allow the connection to the SC2, same as the Disco4G mod.

job2310, 2018-06-12:
For anyone feeling particularly brave, I've put together a bebop2 4G install "package".  The zip can be downloaded from the URL below.  Note the installation for the SC2 is the same as the Disco so please follow those instructions to get your SC2 working.  I've put a few rough install instructions in the "installation.txt" file within the zip.  They should get you started.  Happy to try and help where I can.

The primary differences between the disco4g and bebop4g install processes is that on the bebop2 we've got to kill the /usr/bin/usb_mode process in  order to allow the micro usb to recognise our usb modem.  In addtion, due to the lack of available space on the bebop2, the scripts need to be installed into the /data/ftp/internal_000 folder instead of /data/ftp like on the Disco.

https://drive.google.com/drive/folders/1ecgzVzJHvVKQDevQMu8tItp_3U1C4Rcq?usp=sharing

teja235, 2018-06-12:
I managed to establish a connection but it's basically unusable due to lag and video distortion even though my network speed tests are turning out great. I'm using a Hi-Link stick, does this version incorporate the fix for H modems?

teja235, 2018-06-12:
[image]

job2310, 2018-06-12:
Per the above, I have only tested with 3372-153 modem in Stick Mode.  I do not have a hilink modem.

job2310, 2018-06-12:
Also what version firmware are you running?

teja235, 2018-06-12:
4.4.2, latest firmware for bebop and skycontroller.
I read that you only tested with stick, just dropping a note so everyone can see my results too :) 
Have you incorporated the fix for hilink modems into this code?

job2310, 2018-06-12:
4.7.1 is the latest however this mod doesn't work on that yet.  If you're on 4.4.2 that should be ok.  As I said, my modem converted to stick mode.  So no, hilink modems won't work, at the moment.

ToddTsao, 2018-06-13:
[image]

I am ready for bebop 2 4G.

ToddTsao, 2018-06-13:
Thanks again for Job2310.

uavpal, 2018-06-13:
go have fun and upload some videos! :)

I'll leave the issue open until the code is finalized, fully tested and documentation is ready.

[9 more replies]

closed 2018-06-22 by uavpal