uavpal/disco4g #8: It taking too much time to switch from wifi to 4G mode

dailyknowmore, 2018-05-13:
I just do the mod, find that it taking more than one minute to switch from wifi to 4G.It is really not a good experience and cause problems when flying.
May be the reason is Zerotier.

dailyknowmore, 2018-05-13:
Beside this，there is a problem when i install the software of SKC2. I have to ADB to SKC2 and execute the command line of the installing shell manually.
Excuse my poor english :)

dailyknowmore, 2018-05-13:
Of cause i use same operator in my set. 
In fact the network is a bit complicated in China. Even in same operator, the ip address is NOT in the same LAN(100.x.x.x or 10.x.x.x ) .I have to ping  the IP of another SIM card to comfirm they are in the same LAN.IF they are NOT in the same LAN, I have to reboot one of the device until they are pingable each other.
My operator is China Telecom,May be  i shall write a guide to user in China

dailyknowmore, 2018-05-13:
The sofeware can copy to SKC2  using install shell as plan. But after copy files the rest of the install shell seem not working. i have to execute the rest of shell manually. it's still not esay for common users

uavpal, 2018-05-13:
It worked for other people. Can you screenshot the errors?

uavpal, 2018-05-13:
Your fellow countryman is using China Unicom and gets a direct connection.

dailyknowmore, 2018-05-13:
no error,evey thing seem going as expect,until i double press the setting button of SKC2 and found nothing changed.

I press Ctrl+c when install shell try to ADB to SKC2, then disconnect telnet, telnet disco again, run the install shell and wait the shell ADB to SKC2 finishing the whole installation with no error. May be my intrrupting cause this issuse

I edit it in my cell phone today, so screenshoot is improssible:)

dailyknowmore, 2018-05-13:
I also got direct connection with pptp in my original set.
The ip address not in same LAN is really a problem.China is big with huge population, different area and different operator may have different network circumstance. My network situation may be more complex

uavpal, 2018-05-13:
The IP is never in the same subnet for me, and I still get a direct connection. Also successful ping might no be good indicator whether you will get a direct connection or not. zerotier uses different types of NAT traversal technologies to establish a direct connection.

sami31770, 2018-05-14:
hello please can you explain on video how to transfer disco4g master zip file on filezilla and what to transfer exactly

uavpal, 2018-05-14:
Not any time soon, sorry. but if anyone else is keen on doing so, I'm happy to link to the video in the instructions.

sami31770, 2018-05-14:
Thank you for your answer
if i added the softmod on disco  with Filezilla i don't need to do this procedure with command promp?? next here++

Open a Command Prompt (Windows)/Terminal (macOS/Linux) and Copy/Paste the following commands manually to install the softmod. 
telnet 192.168.42.1 (If you get an error under Windows, make sure the Telnet Client is installed)
mv /data/ftp/disco4g-master /tmp/disco4g
chmod +x /tmp/disco4g/*/*_install.sh
/tmp/disco4g/disco/disco_install.sh
/tmp/disco4g/skycontroller2/skycontroller2_install.sh
reboot

uavpal, 2018-05-15:
These are step by steps instructions. In other words: you need to do every single step.
I'm closing this issue as the recent messages have nothing to do anymore with the initial issue topic ny dailyknowmore.

closed 2018-05-15 by uavpal