uavpal/disco4g #9: Modems

nemesis668, 2018-05-13:
Hey buddy, Lots of people asking in UK which modem is right me included, Would rather not have to order from china and wait a month :)

Does this modem seem to be the correct one?

https://www.amazon.co.uk/gp/product/B07BNRZH3T/ref=ox_sc_act_title_1?smid=A1DWT37M9SCX7N&psc=1

dailyknowmore, 2018-05-13:
I use ZTE mf825C, the configuation is easier than huawei 3372.  Just setting the  route and dns server ,the CHUNCK can surf  internet.
I think ZTE mf823, ZTE mf831 should work in the same way

uavpal, 2018-05-14:
The Amazon link does not specify the type (H vs. S) and the model number (e.g. 153, 210, 510, 607).
You probably want to buy a E3372s-153 for the UK.
You can try ebay if shipping from China is not an option.

sami31770, 2018-05-14:
Hello uvapal can you please explain on a video step by step how you connect the modem and the disco with windows please and how to transfer the file to the disco please

sami31770, 2018-05-14:
i recently buy huawei E3372h-607 do it work in europe

uavpal, 2018-05-14:
please see https://github.com/uavpal/disco4g/wiki/FAQ#e3372models

closed 2018-05-15 by uavpal