uavpal/disco4g #10: Sky Controller 2 Not Showing In ZeroTier

nemesis668, 2018-05-15:
So Issue I am having is the sky controller 2 in not showing in zero tier, It's conencted to my phone as that shows 1 device connected to hotspot, The sky controller is flashing blue. My disco shows in zero tier, but my sky controller does not . Any ideas?

nemesis668, 2018-05-15:
Also Another thing I am not getting any sms message through from glympse

teja235, 2018-05-15:
Try connecting another laptop or some device to your hotspot to check the internet connection. Maybe your hotspot isn’t working?

uavpal, 2018-05-16:
If you have a data only SIM in your disco, it might not support outgoing SMS. You can still use the map with real-time location inside FreeFlight Pro. Glympse is not a must-have.

double check that skycontroller2/uavpal/conf/zt_networkid is set to the same as in disco/uavpal/conf/zt_networkid

uavpal, 2018-05-29:
nemesis668, is this resolved?

nemesis668, 2018-05-29:
Yes thank you

jakecobey, 2018-09-03:
I also have the same problem... the st I’d is identical, hotspot connects on my phone ok - iPhone X and on o2 network...

The disco show online in zt but nothing from sc and blue flashing light.... any ideas?

uavpal, 2018-09-04:
@jakecobey please join us on https://uavpal.com/slack, will help to troubleshoot there