uavpal/disco4g #11: Did you get a good better experience flying with antenna?

dailyknowmore, 2018-05-17:
Beacuse of missing relay server of Zerotier in my home country, I  remain trying my own solution(nat + pptp). I got a not bad experience in test (the SKC2 stay at home, I drove  about two kilometer away with  Disco using a car). But it is totally disaster when flying, although it behave well in ground. It disconnect in ten seconds after takeoff. reconnect then  disconnect soon.
Now I consider to install antennas. I dont think it can improve the signal originally, but now I have to try it.
Beside this, I also consider to do a Zerotier moon server to improve the experience of uavpal solution, or trying another nattraversal solution 
Can u give me some advice? thanks in advance

uavpal, 2018-05-29:
dailyknowmore, can I close this issue?

teja235, 2018-05-31:
Cell tower signals are designed for use at ground level. Antennas and boosters could help clean up the signal, but for best results you wanna stay closer to the ground. Also while I commend your efforts, I have to note that while testing a signal over the internet, the distance between the drone and the controller makes no difference whatsoever (kinda the whole point of the mod). You didn’t really have to drive anywhere, But I’m glad you’re working on it and making progress! Let me know if you discover something, maybe there’s better solutions than ZeroTier.