uavpal/disco4g #13: SKC can not connect Disco via zerotier

stishsun, 2018-05-23:
I have successful connect Zerotier both of Disco and SKC. the zeiotire shown both device in list.

I also can ping the peer IP address assigned by Zerotier from Disco.

but the SKC still flash blue not change to steady blue...

any idea?

uavpal, 2018-05-24:
Most probably, because Ethernet Bridge is not enabled for **both** devices in Zerotier....?

stishsun, 2018-05-25:
yes, you are right, finally I found the icon to bridge them

thank you

uavpal, 2018-05-25:
Thanks. I added it to the FAQ:
https://uavpal.com/faq#blueflashing

Closing the issue now.

closed 2018-05-25 by uavpal

stishsun, 2018-05-25:
this icon is "wrench", not "screwdriver", I looking fo "screwdriver" icon, not not found...

uavpal, 2018-05-25:
lol, sorry. You're absolutely right! I just corrected the installation wiki and FAQ. Thanks!