uavpal/disco4g #14: how to use Glympse?

stishsun, 2018-05-29:
how can I setup my App to retrieve Disco position shown in App?

uavpal, 2018-05-29:
1. Install the app on your mobile phone
iOS AppStore: http://phobos.apple.com/WebObjects/MZStore.woa/wa/viewSoftware?id=330316698&mt=8
Android Google Play: https://play.google.com/store/apps/details?id=com.glympse.android.glympse
Windows Store: http://www.windowsphone.com/en-us/store/app/glympse/ec418e52-b788-e011-986b-78e7d1fa76f8

2. Just open the link received via SMS in Glympse. The tracking should start automatically.

stishsun, 2018-05-29:
how the APPs knows Disco position? there should have some linkage between Glympse Apps and Disco.
Glympse using phone number and SMS to login, the phone number in Disco Config is same your phone number?

uavpal, 2018-05-29:
The softmod running on the Disco is constantly reading out the GPS coordinates from CHUCK‘s GPS sensor. The location is forwarded to Glympse (web service) via API. At the same time a unique Glympse address is created and sent to the phone number specified in the config file via SMS.
Anyone with that (private) link can track your Disco for 4 hours.
I‘m closing this as it‘s not an issue. I suggest you ask questions on Slack if you need further clarfication. Thanks.

closed 2018-05-29 by uavpal

stishsun, 2018-05-30:
Thanks your clarify. the phone number config in Disco is for the Phone which will receive the Disco location, This phone number is not Disco's phone number. if this is correct, it should be included in the description to prevent confusing.

uavpal, 2018-05-30:
Sorry, but RTM "to prevent confusing" :)

Have a look at the config file: https://github.com/uavpal/disco4g/blob/master/disco/uavpal/conf/phonenumber
The description/comment clearly says: "This is usually your mobile phone number and **not the number of the SIM inside the disco.**"

and have a look at the FAQ:
https://github.com/uavpal/disco4g/wiki/FAQ#nosms
"The phone number in conf/phonenumber should be the device where you want to receive the SMS with the link (**and not the Disco's SIM phone number itself**)."