uavpal/disco4g #15: Disco FW 1.7.0 Compatibility

teja235, 2018-05-30:
Hey Guys!

Parrot issued a firmware update a little while ago, so I thought I'd update and see what happens. I did a factory restore of the drone by double clicking the power button and holding it until the colors flashed. The new firmware requires you to double click on the power to activate telnet, but once I did that and completed the install, it's still not working. Zerotier shows both devices have connected to its network, but they are not connecting to each other. Any ideas?

teja235, 2018-05-30:
Does disco4g output some kind of log to see what's going on? I don't understand why they don't see each other if they've connected to the network! I feel like that was the hard part >.<

job2310, 2018-05-30:
Have you ticked the box to allow Ethernet bridging on both devices in ZeroTier?  Can you ping the zerotier address from one to the other?  Does zerotier show both as online?

teja235, 2018-05-30:
Yes Ethernet bridging is active, and zerotier shows both devices connected in green. When you say ping, you mean to ping SC2 from the disco via telnet right? Would I ping the private ip assigned by zerotier?

job2310, 2018-05-30:
Correct, telnet into the disco and ping the zerotier IP of the SC2.  Also try the command below on the disco. Do you see the unique id of the SC2 as assigned by zerotier in the output?

zerotier-one -q listpeers

teja235, 2018-05-30:
The pings are not being responded to. I tried pinging both the public and private addresses of SC2, no go. Then I connected my phone to the zerotier network as a VPN, and tried to ping that but still no go. 
The command you gave me is not recognized.

[image]

job2310, 2018-05-30:
Try...

 /data/ftp/zerotier-one -q listpeers

teja235, 2018-05-30:
[image]

teja235, 2018-05-30:
Found it!

[image]

What does this mean? lol

teja235, 2018-05-30:
OH 0a4a94c5bf is the SC2. I don't recognize any of the other ones, except the last one is part of my network ID

job2310, 2018-05-30:
Is one of the ztaddr your SC2?

teja235, 2018-05-30:
Yes

job2310, 2018-05-30:
Yeah I’m not to sure.  I’ve not done the update yet so can’t confirm if that’s really the issue here.  Just to confirm you’re able to ping google.com or something else from the Disco and SC2?

teja235, 2018-05-30:
Yup, able to ping other sites just fine from the disco. not sure how to ping from SC2, but Zerotier shows both devices are connected to the network. The disco is able to see the SC2 when running /data/ftp/uavpal/bin/zerotier-one -q listpeers, and it disappears from the list when the SC2 gives up and switches back to wifi mode.

tldr; everything is working fine except the connection. Does anyone know if it's possible to downgrade the firmware? I tried googling it but all the videos are in french :/

uavpal, 2018-05-31:
I will check it out tonight. If there's no way to get it to work short term, I will publish downgrade instructions.
Have you tried reinstalling the softmod?

uavpal, 2018-05-31:
Ok, thanks. I will do a deep-dive tonight. Everyone else: do not update for now.
SoftIce

uavpal, 2018-05-31:
I played around with 1.7.0, I don't see an issue to make the 4G softmod compatible with it, but it will take some time as I need to cross-compile most kernel modules again.

People who accidentally upgraded and can't wait for the 4G softmod fix to be out: this is how you downgrade back to 1.4.1 again:
https://github.com/uavpal/disco4g/wiki/Downgrade-Parrot-Disco-Firmware-1.7.0-to-1.4.1

andyjenkinson, 2018-05-31:
Possibly a security update if they bothered to update the kernel version?

To be fair, the telnet change is welcome from one perspective. Parrot drones have been seriously exploitable for years. Defaulting to an open WiFi network, with unauthenticated telnet access (as root). Not sure if much has changed since the AR.Drone days but it was always possible to telnet in, kill the pilot software to cause it to fall out of the sky, or just take control (the controls being an API after all). I notice at least that the WiFi and telnet appear to be unchanged (until now).

uavpal, 2018-06-04:
I got it working under 1.7.0. Due to people reporting various issues (not 4G softmod related) on 1.7.0, I will not yet update the code and recommendation on GitHub for now.

If anybody wants the kernel modules for 1.7.0, please PM me on Slack, happy to share. For now, recommendation is to downgrade (https://github.com/uavpal/disco4g/wiki/Downgrade-Parrot-Disco-Firmware-1.7.0-to-1.4.1) and install the softmod on 1.4.1.

@andyjenkinson the wifi connection is usually secured via WPA2 (auth+encryption), therefore I don't see the need to disable telnet by default as only the disco and SC2 should be part of that WLAN. However, I had to implement iptables for the softmod to block incoming telnet/ftp/... connections via 4G, which in case of public IP is a potential 2^32 attackers :)

Rgds, SoftIce

stishsun, 2018-06-04:
Thanks, I would like to share my experience in Disco4G, in some case, if the coverage is no good and Disco disconnect. the SC would still stay steady blue or flash green, from the sc I am not sure what is current status., btw, if the 4G disconnect, Disco will RTH and circle land, no longer circle hover in Home point.

stishsun, 2018-06-04:
I have not yet upgrade to fw1.7, all of my experience is v1.41, it seems the mod of sc is not so good as disco

uavpal, 2018-06-04:
@stishsun, please open a new issue (if there is one?) or join us on Slack if you need help.
Let's not spam this thread, as it is meant for the purpose of "Disco FW 1.7.0 NOT COMPATIBLE".

uavpal, 2018-06-04:
The softmod code and kernel modules have been updated and now fully support the latest two Disco Firmware versions 1.4.1 and 1.7.0.

Closing the issue hereby.
SoftIce

closed 2018-06-04 by uavpal

uavpal, 2018-06-05:
https://github.com/uavpal/disco4g/blob/master/README.md#contactcontribute

cheetah2k, 2018-06-28:
Hi guys,  on 1.7.0 and I am still having the issue with the Disco not connecting to Zerotier..  help!

job2310, 2018-06-28:
@cheetah2k 
Is your modem making a connection to the internet? (solid blue light)
What is the output from running ifconfig

javieryine, 2018-07-02:
Hola soy de Uruguay gracias por este gran aporte softmod bueno el disco se conecta a zerotier luz verde pero skycontrol se conectó una ves y después no se conectó más  luz de skycontrol en color púrpura flayeando y no se conecta a mi celular hospod alguna ayuda?

uavpal, 2018-07-02:
English please.
Have you tried https://uavpal.com/disco/faq#wififail ?

cheetah2k, 2018-07-02:
Hey guys I sorted mine out - just had to full uninstall and reinstall :)

javieryine, 2018-07-02:
I'll try thanks