uavpal/disco4g #17: ZeroTier does not see the Disco or the SC2

richfriz, 2018-06-05:
I am probably doing something stupid butt... I can get everything working except no devices show up on ZeroTeir. I have tried making new networks, changing the network id in the conf files and uploading again. I have done this about 5 times. On one occasion I did see the Disco but not the SC2 ?? The SC2 connects to the hotspot. The modem is working (solid cyan LED) I meticulously cut and paste the network id so there should be no mistake.  I also ran the "downgrade" mod. All to no avail 

[image]
[image]

uavpal, 2018-06-05:
can you telnet into your disco and tell us what `ulogcat |grep uavpal` returns?

uavpal, 2018-06-05:
feel free to join us on slack (https://github.com/uavpal/disco4g/blob/master/README.md#contactcontribute) and PM me, so you don't have to post your logs all over the Internet :)

richfriz, 2018-06-06:
All solved : Thank you for the fantastic support

closed 2018-06-06 by richfriz