uavpal/disco4g #18: Skycontroller 2 don't see updates + manual firmware update not working

olrac86, 2018-06-06:
Hello, 

I'm installing uavpal soft mod on my Disco. 
After the installation phase, everything seems to work well with the drone (I can see it on ZeroTier).
The problem is with my SkyController 2 (SC2). The installation phase seems not to work, so I did little investigation... and I discovered that my SC2 has still 1.0.3 firmware. One problem is that freeflighPRO doesn't recognise any updates, it says that 1.0.3 is updated (which is false, since the last firmware of the SkyController 2 is the 1.0.7).
So I tried to do the manual SC2 firmware update by following your wiki, but it doesn't seem to update properly: I've directly the 1.0.7 .plf, which goes to segmentation fault; then I tried the 1.0.4 .plf, which doesn't segfault, but it seems to freeze the shell, and nothing happens. 

Is there anything I can try ?

uavpal, 2018-06-06:
Duplicate of https://github.com/uavpal/disco4g/issues/16
@ryangoldkey reported the same issue. He also got segmentation fault trying to upgrade. We tried a few more things but ran out of ideas eventually. It works fine for me though.

I closed the earlier GItHub issue (#16) as it's not really a Disco4G issue. The softmod is compatible with SC2 firmware 1.0.7 only.

Is you Skycontroller a 2 or 2P (see sticker below)?

There is one more idea I have: are you using Android or iOS for FreeFlight Pro? Parrot does not package firmware updates into the Android Apps, but they do for iOS Apps.
In other words, the SC2 firmware 1.0.7 is part of the latest iOS FreeFlight Pro (v5.2.0). If yours is Android, try to get you hands on an iPhone/iPad, install FreeFlight Pro and plug the USB cable into SC2 - does it recognize the update then?

If not, you need to take it up with Parrot Support...

Good luck!
SoftIce

olrac86, 2018-06-06:
It's a SkyController 2P. 
I've already tried iOS FreeFlight Pro (v5.2.0) :***(

olrac86, 2018-06-06:
Btw, I have one last question. Do you see a real reason why the 4G soft mod will not work with firmware 1.0.3 ? Do I have any chance of throubleshooting it by myself or is it just a no go ?  
Thanks

olrac86, 2018-06-06:
I've started throubleshooting and came to the conclusion that it really seems nothing can be done to make the soft mod running on the SkyController2P with firmware 1.0.3 (at least, without recompiling everything).

Still, this activity gave rise to the following thought in my mind (correct me if I am wrong):
theoretically, we don't need the SkyController2 for real, as FreeFlightPro on iOS should be enough to pilot the Disco (eventually, adding an USB joypad controller). I mean, a tablet/smartphone with wifi and 4G connection should be enough for talking, on one side to ZeroTier -directly-, and on the other side with the SkyController2 --as if the tablet/smartphone were the Disco. 
Stated otherwise, the tablet/smartphone can be the only Man in the Middle.
This should be theoretically feasible. I will try...

uavpal, 2018-06-07:
Question: are you at least able to download the SC2 firmware in FreeFlight Pro?

uavpal, 2018-06-07:
The answer is very simple:

- Latest firmware for Skycontroller **2** (internal hardware identifier: 090f/mpp) is 1.0.7
- (build date: Oct 24 2017)
- Latest firmware for Skycontroller **2P** (internal hardware identifier: 0915/mpp2) is 1.0.3
- (build date: Jan 25 2018)

Distinguishing characteristic: the Skycontroller 2P has black joysticks.

In other words:

- SC2P with 1.0.3 is even newer than SC2 with 1.0.7.
- These are two different hardware platforms and the firmware files are not interchangable between SC2 and SC2P.
- The version 1.0.3 on SC2 is not comparable to version 1.0.3 on SC2P.

olrac86, 2018-06-07:
Oh, I see! That's a nice discovery, good to know.
Well, this means there might be hope for porting the soft mod from SC2 1.0.7 to SC2P 1.0.3.
I remain at your disposal, SoftIce, in case you want to try any idea on my SC2P platform.
For what I understand, there's nothing to recompile, so let me try to re-run /data/lib/ftp/uavpal/bin/uavpal_sc2.sh manually and see again where it fails...

ryangoldkey, 2018-06-07:
Dear olrac86,

pls let us know about the outcome you got on this. I'm already in contact with parrot regarding exchanging the controller. If you find a way to run on 1.0.3. pls let me know.

thanks in advance and best regards!

cyberjunkiefr, 2018-07-23:
Hello olrac86,
I'm looking for a friend who has a sc2p.
I found your script but I don't know how to use it. Do I just replace the file in the Dusco4g_master.zip files with yours and then do the original installation procedure?

Thanks for your help

Roel

radisc, 2018-08-04:
Hi, 
same question of cyberjunkiefr.

Nicola

uavpal, 2018-08-04:
Yes, just simply copy it over the extracted skycontroller2/uavpal/bin/uavpal_sc2.sh before running the installer steps in the manual. If you need help, join us on Slack.
This will be the procedure until I merged the code.

VicJTChen, 2018-09-04:
Hi, I downloaded the script, overwrite this file to the original one and follow the procedure to install the code (https://github.com/uavpal/beboptwo4g/wiki/Installation). However, my SC2P still can't connect to the wifi hotspot, it still connects to my drone directly. What step else do I need to do further??

uavpal, 2018-09-04:
The script works fine for the Disco, but not yet for Bebop2. Can you try to manually replace all
"DISCO" with "Bebop2" in the script?

olrac86, 2018-09-04:
I've already tested it on the bebop2 and it works.
As softice suggets, just replace "Disco" with "Bebop2" (or the correct
system name of your Bebop2 machine) in the script.

BadWolf42, 2018-09-09:
Hello @uavpal, olrac86,

As discused en Slack, I merged SC2P code in master here : https://github.com/uavpal/disco4g/compare/master...BadWolf42:BadWolf42-sc2p
The two main SC2 and SC2P scripts are provided with the package.
Detection of sc2 or sc2p is done at installation, if product in build.prop on controler is mmp2 (sc2p), uavpal_sc2.sh is replaced by uavpal_sc2p.sh otherwise uavpal_sc2p.sh is deleted before push to the SC.
I modified native the uavpal_sc2.sh and the uavpal_sc2p.sh provided my olrac86 to match coding style and benefit one from another.

Hope it helps,
Bad

BadWolf42, 2018-09-10:
Sure the main goal is to merge all in one file.
Currently I had to merge it in disco or in bebop2 as it's a more important feature than the global renaming, but no I don't think it's Disco or Bebop2 specific the code is pretty much the same for the SC between the 2 platforms, isn't it ?

Is you consider to have a merged repository for the 2 platforms soon, I can directly work on a merge revision, but some portions of the code here are still a little bit obscure and I need to better understand why they are here for.

Bad

uavpal, 2018-09-10:
Olrac's script has some Disco/BB2 specific code (i.e. hotspot SSID prefix), see https://github.com/uavpal/disco4g/issues/18#issuecomment-418216764

BadWolf42, 2018-09-10:
Ok thanks for pointing that out, I'll take a look at a was to generalize that part.

Helderf33, 2018-10-06:
Hi uavpal. I'm getting an error when trying to get your mod working on my bebop2, after the reboot command on telnet.

/ # reboot
POMPCLI: event_cb : event=0(CONNECTED) conn=0x1292918 msg=(nil)
POMPCLI: event_cb : event=1(DISCONNECTED) conn=0x1292918 msg=(nil)
umount: can't unmount /data: Device or resource busy

After reboot nothing happens, router does not have light always on.
I have the SC2P is that why this error is happening?

Thanks

uavpal, 2018-10-07:
Did you follow the manual steps for the SC2P in the comments above? (https://github.com/uavpal/disco4g/issues/18#issuecomment-410447307)

... and what router are you talking about?

I suggest you join the community on https://uavpal.com/slack

SoftIce

Helderf33, 2018-10-07:
The router is the one you suggest and I checked the version it is the right one for my country.
I did follow the steps but my technical knowledge is very limited so I trying to figure out if I screw up altering the file that olrac86 provided.
I will make a slack account an join you.
Thanks

uavpal, 2018-10-07:
router: you mean modem :)
no need to alter the script, just replace it.
see you on Slack! :)

Helderf33, 2018-10-07:
Lol, yes that pen thingy :)
I don't have to alter it for the disco but I do for the bebop2 or did I get that wrong?
I tried with the file as is provided but did not work without the changes either.

olrac86, 2018-11-29:
Hi, I've tested my code with the new SC2P firmware, 1.0.5, and it works out of the box. It needs a re-install, though.

s0ftice, 2018-12-30:
I merged @olrac86's code with the existing SC2 uavpal_sc2.sh script. Tried to re-use as much from both scripts (e.g. jstest no longer required).

The script can be downloaded from the master (currently v1.2-beta) directly: https://raw.githubusercontent.com/uavpal/disco4g/master/skycontroller2/uavpal/bin/uavpal_sc2.sh
for users who only want to upgrade the controller but not the drone.

This does not require any changes and supports Disco/Bebop2 out of the box.

I will finally close this issue once v1.2 is released.
Special thanks to @olrac86 and @BadWolf42 !!!

SoftIce

s0ftice, 2019-01-08:
v1.2 has just been released. This issue is therefore closed. Thanks again to everyone who contributed!

closed 2019-01-08 by s0ftice

olrac86, 2019-01-08:
Thanks to you, Marc!! Happy New Year!

Cheers!