uavpal/disco4g #19: Can't get Glympse developer account, also no video stream

radisc, 2018-06-11:
Hello, i installed the mod yesterday, but i wasn't able to get a glympse account, as it has been weeks that when i click in the verification link in the email it keeps telling me: "An unexpected error occurred while processing your verification request. Please try again later. "
I understand it is optional, but still a bit frustrating.

Another issue, after setting everything up, zerotier says both SC2 and disco4g are online, freeflight seems to connect, but once connected the video is completely black, additionaly, the bandwidth meter on my phone shows around 400KB/s, as if the video is actually being transmitted, and i can start the main engine.

Thanks for the assistance and great work guys, for real

uavpal, 2018-06-11:
Hi! I guess you will need to setup another Glympse account with a different e-mail address. I once tried to delete a test account, but there is also no way of disabling/canceling accounts to my knowledge.

A bad video connection can be caused by various reasons. I suggest you join us on Slack (https://uavpal.com/slack) and describe your setup in detail (e.g. carrier, APNs used, ...)

Rgds, SoftIce

uavpal, 2018-06-22:
radisc, is this still a problem? Other people were setting up new Glympse accounts the past few days and no issues reported.

I will close it, as it's not an Disco4G issue.
SoftIce

closed 2018-06-22 by uavpal