uavpal/disco4g #23: mod not working correctly

arthur9711, 2018-06-28:
Hi,

first thanks for everything.
But i have a big issue
at first all was working but since this morning nothing works.

i've tried uninstall and reinstall but the disco doesn't show a blue flashing led whereas the modem is correctly connected and showing a cyan led.

i don't know from where the problem come from.

the skycontr is showing a blue led and correctly connect to my iPhone.

on zerotier both are connected .

arthur9711, 2018-06-28:
After the last uninstall i've made and nothing plugged into disco and skycontr the led on disco flash blue ...

arthur9711, 2018-06-30:
@uavpal maybe you've already seen this issue ?

uavpal, 2018-07-03:
I will close this issue if I don't hear back.

closed 2018-07-22 by uavpal