uavpal/disco4g #25: ZeroTier connection

Thyristornp, 2018-07-09:
Hello!

I did everything according to the instructions.
I have Authorize Disco and Skycontroller in ZeroTier, but always when I turn they on, I see only one member online, Disco or Skycontroller or nothing.
They connects only sometimes.
USB modem is OK and Skycontroller is connected to my phone via Wi-Fi.
Maybe it is IP blacklisting in Russia? How can I check this connection to ZeroTier? May be it is a command like a ping?

uavpal, 2018-07-09:
you can run `/data/ftp/uavpal/bin/zerotier-one -q status` to check its status

Thyristornp, 2018-07-09:
I have this message:

/ # /data/ftp/uavpal/bin/zerotier-one -q status
200 info 460e4f9dde 1.2.6 ONLINE

But, there is no connection in https://my.zerotier.com/network

uavpal, 2018-07-10:
he status in the Web UI is not always accurate (however most of the time it is). Can you see an IPv4 address assigned to the members? Can they ping each other? Do you get steady blue LED on the controller?

uavpal, 2018-07-10:
did you ever delete a member for a network ID in the zerotier UI?

Can you join the community on https://uavpal.com/slack and upload a screenshot of your zerotier account (network tab)?
Thanks,
SoftIce

uavpal, 2018-07-22:
can I close this issue?

ct1ftk, 2018-07-22:
I have already installed everything and configured it as described and I am almost giving up on this solution because I can not establish communication with "ZEROTIER" (between Disk and SC2). I have spent many hours in experiments and tests, I have spent too many hours of rest. I'm saturated!
Please, anybody can help me?
Thanks,
Henrique Manso

ct1ftk, 2018-07-22:
See the screenshot in attach:
[image]

[image]
[image]

uavpal, 2018-08-03:
all resolved now? can I close this issue?

closed 2018-08-04 by uavpal

s0ftice, 2018-12-21:
please ask on slack in #Disco4G for help

mysaidi007, 2018-12-21:
Do you have the link?

s0ftice, 2018-12-21:
https://uavpal.com/slack

mysaidi007, 2018-12-21:
Already join found nothing

mysaidi007, 2018-12-21:
Can you give me the link direct to the disco 4g issue

s0ftice, 2018-12-21:
If you don't ask for help on Slack, people wouldn't know you need some :)

kimkabsoo, 2019-02-04:
HI !
I have bebop2

Blue LED on the controller is blinking blue after connection to phone wifi, but I can't Authorize it

I have this telnet message:
/ # /data/ftp/uavpal/bin/zerotier-one -q status

/data/ftp/uavpal/bin/zerotier-one: missing port and zerotier-one.port not found in /var/lib/zerotier-one

s0ftice, 2019-02-04:
Please join us on Slack for installation troubleshooting