uavpal/disco4g #27: Hi link vs Stick mode

BROLLIC, 2018-07-14:
Hello all. I have a hi link usb stick. Everything connects (It can take anywhere from 20 seconds to 3 minutes... I have a great 4g connection as well). 95% of the time after its all connected and the controller light is solid blue, it will randomly disconnect at random times. (Flash red) Sometimes it will reconnect, sometimes it wont and the craft has to RTH. I know my area has great 4G so Im assuming it isn't that. Do you think it could be because I have a Hi link 4g stick instead of a stick mode one? Is there anyone here using hi link with stable connections long enough to have a successful long distance flight? What do you think I should do?

teja235, 2018-07-14:
I also use apps like Sensorly and OpenSignal to see where the nearest tower is and position myself accordingly

BROLLIC, 2018-07-14:
Ahhh ok. Im getting anywhere between 5-15mbs down and 10-20mbs up. Also tmobile. NYC. What do you think is acceptable speeds for a successful flight.

teja235, 2018-07-14:
The video bitrate at 720p is 15-20 mbps. At highest quality it can hit up to 40 mbps. The additional issue is that when there’s limited bandwidth, your phone will prioritize its own traffic over hotspots so make sure you close all apps and background refresh and all that stuff. Finally, T-Mobile won’t give you LTE speeds on personal hotspot unless you have a hotspot plan. I have TMobile One Plus, 10 GB hotspot for $10 extra. Also the usual caveat of in order to get a direct connection, both the sim in the disco hotspot and your phone hotspot have to be T-Mobile lte on the fast.tmobile.com APN. I have a gigabit connection at home with 700+ d/u and I still haven’t managed to get a usable connection. That’s just how it works lol, unless maybe we could all pitch in and get an AWS server running 😂

BROLLIC, 2018-07-14:
Um, what state do you live in lol? Most states dont get 70mbps down lol

uavpal, 2018-07-15:
See https://uavpal.com/disco/faq#datausage
the video bitrate required is actually much less, @teja235 where did you get these figures from? (20 mbps is even good enough for streaming 4k in decent quality...) :)

Disco/Bebop2:
480p ~2.4 Mbps
720p ~4.8 Mbps

@BROLLIC I suggest you join the community on https://uavpal.com/slack

SoftIce

closed 2018-07-22 by uavpal