uavpal/disco4g #28: no conection between controller and disco

sully2023, 2018-07-14:
setup everything as per instructions...both disco and controller show up on zerotier,have authorized both and enabled bridging on both but the controller remains on flashing blue light until it eventually gives up and returns to green wifi conection,have tried everything but just dosnt seem to make any difference...anyone able to help

uavpal, 2018-07-15:
you can try to install zerotier on a PC and add it to the same network. Once disco and SC2 are online, try to ping both your PC using the zerotier IP addresses. If both are up and running, try to telnet to 192.168.42.1 (you might need to manually assign a 192.168.42.x IP to your PC too)

uavpal, 2018-07-22:
can I close this issue?

ct1ftk, 2018-07-22:
I have the some problem. I have already installed everything and configured it as described and I am almost giving up on this solution because I can not establish communication with "ZEROTIER" (between Disk and SC2). I have spent many hours in experiments and tests, I have spent too many hours of rest. I'm saturated!

uavpal, 2018-07-22:
Join the community on https://uavpal.com/slack and post a screenshot of the members in your zerotier web UI, or PM me.

SoftIce

sully2023, 2018-07-22:
Yes mate,you can close this,I completely give up,spent to much time trying to get this to work and not enough flying,uninstalled,reinstalled just got worse,controller stopped working with upgrade then worked so that’s it....if anyone does a step by step video from beginning to end then I might have another go but until then

closed 2018-07-22 by sully2023