uavpal/disco4g #31: Pixelated/ Not clear video feed

BROLLIC, 2018-08-01:
Hello, I have successfully installed the softmod on both my disco and bebop 2. I can get a solid blue light connection on both anywhere between 10-30 seconds. I can operate it fine (flight controls) The only problem (MAJOR) is that the live video feed is pixelated. So I cant properly see where Im going. I actually went 100 feet from my US T-mobile tower and tested its speed. (35mbps down, 40mbps up)....so I know I have enough speed. Could it be my modem? I have the 510 Hi Link modem, I cant find any stick mode ones :(. I have also tried to configure the video feed in the parrot app. I have used android and apple devices. They fly perfectly, no lag in controls at all. No matter how high or how low I go, the video is still not viewable. Please help :)

teja235, 2018-08-01:
Do you have T-Mobile One+ ? You need to have a hotspot plan on your phone line in order for the controller to get the full bandwidth your phone has. It only costs an extra $10 /month if you already have T-Mobile One. Also the sim in your Huawei must be the same as your phone (Tmobile), and must has a data plan that works for hotspot devices. I have the 10 gb /month plan. Gets me about 11 hours of 4G flight

BROLLIC, 2018-08-01:
Yes I have all the Tmobile bells and whistles! Ive tested my phones hotspot speeds with 3 other tablets/phones. I can get 25 mbps down and 30 mbps up. So what else could it be :(?

teja235, 2018-08-01:
Yea, test the speed and latency of your phone hotspot from a different
device. Then test the huawei speed and latency by plugging it into a laptop
or something. If everything checks out, check your settings on zerotier.
Ethernet bridging should be checked.

If all this doesn’t fix the issue, maybe it’s just the quality of signal at
the spot you’re taking off from. Cell signal is just another radio signal
and susceptible to noise from obstacles and reflections. Try a different
spot! I had to try a couple different ones before I could find one with
direct line of sight to a tower and quiet enough to fly without people
asking questions of legality 😅 use the apps Sensorly and OpenSignal to
find your nearest tower and measure the signal and speed.

BROLLIC, 2018-08-01:
I did all of that lol. I was litterally 100 feet from a tower... If could have sat on it if i wanted to but i thought that might be overkill lol. Still wont work. I tries multiple towers here in NY. Can I purchase a stick mode modem from yall? Ive seen some uavpal signed modems out there and would love to get one as well.

teja235, 2018-08-02:
If you have the latest version of the hack installed, it shouldn’t matter.
I have an h modem on T-Mobile and I get a great experience. My plane can
even fly in places where my phone doesn’t get signal, presumably because it
flies over obstacles and valleys, getting a better reception. Maybe
T-Mobile New York just can’t handle the load 😓

teja235, 2018-08-02:
Just have the make sure my phone is in a good spot to get signal and make a
strong hotspot. Also close all the apps you can and let the phone focus on
being a hotspot. I put it up in a tree like an antenna.

BROLLIC, 2018-08-03:
Ok everyone. I finally called t mobile up. I explained my situation. They told me that I did purchase the one "PLUS" package, but that there was an error on their side on activating it. So now I have full speed data on my modem attatched to the drones.

So if you ever have video issues but can operate the birds fine, hit up your phone service provider and question your internet speeds! Video's from NY coming soon.

teja235, 2018-08-03:
Yup, I always question the phone hotspot first because that’s the one
that’s most likely to fail. Glad you got it figured out! Have fun and be
safe, flying near major cities can be risky. Watch out for helicopter
tours!

closed 2018-08-03 by uavpal