uavpal/disco4g #33: Rth

biky83, 2018-08-06:
Hello, I’ve lost the connection (45 secondes)
The disco gone at the opposite of my position. No circle, the magnetometer looks good and the drone distance is ok with my position. When the connection was ok, I make a U turn to go back. 
Does the rth really works ?

teja235, 2018-08-06:
It does when there’s a complete disconnect. The problem is sometimes over
4G, the video connection goes off but telemetry and controls are still
active so the drone doesn’t register the disconnect.

Whenever I have an issue like that, I immediately switch back to Wifi. That
will ensure the drone will begin its journey back and will have a
connection once it’s back in range.

biky83, 2018-08-06:
On the sc, I had the message « not  connected » in red. The next time, I will try to enable the wifi.

uavpal, 2018-08-07:
You can also try to click the Back button in FreeFlight Pro and go into the fullscreen video again to re-establish video connection.

biky83, 2018-08-15:
Yes but I have a problem with 4g connection
The stick is connected on lte but there is a deconnexion with the sc. I have tested with the parrrot near me and my phone is connected on lte. The huawei has a better reception level.

And if the connection is re established, I have a huge latency (7-8 secondes)
The connection between the sc and my phone is very long, sometimes it connected after the second try.
I have the same carrier on the phone and the stick, and have downgrade the stick to stick mode (was on hi link before)

And my connection and speed is better without antenna, I order an lte patch antenna mimo commpatible to test it.

Last question, what the disco does when the connection « migrate » to another carrier antenna ?

can I close this issue?

uavpal, 2018-08-15:
There is usually no roaming between carriers within their respective home country. Therefore the disco will only connect to another antenna of the same carrier when flying out of an antenna's coverage.

What carrier are you using? For France, Bouygues and Orange seem to work well. I will close the issue, feel free to join us on Slack if you have more questions, there are more people from France, in case you have carrier-specific questions.

SoftIce

closed 2018-08-15 by uavpal

biky83, 2018-08-15:
I’m using orange
I just download slack but the app ask an invitation to join a workspace

Closed #33<https://github.com/uavpal/disco4g/issues/33>.

uavpal, 2018-08-15:
https://uavpal.com/slack