uavpal/disco4g #34: Removing Mod / Changing Configuration

szorba, 2018-08-06:
Hi,

i just installed the mod via telnet. Unfortunately i inserted a wrong zerotier id into the conf-file. How do i get this corrected ? I realized that during the installation a routine copied files onto the skycontroller so just editing the file via ftp wont help because the installation routine has to be started i guess.

The instructions to remove the mod which i found in the wiki does not work, it looks to me like an installation routine, not a de-installation procedure because telnet tells me it cant find the directories.

Greetings
Serdar

highwind001, 2018-08-07:
Hi, i had this problem but for bb2. Telnet to the drone, right click the file internal_000, and delete. Right click the updated disco4g folder and upload again, basically reinstalling.

uavpal, 2018-08-07:
@highwind001 removing internal_000 does not remove everything, I don't recommend to do that. Not even sure if the Bebop2 re-creates it, it is required to record the videos... On the Disco, we don't store the files in there, so it won't help @szorba assuming he is doing it for a Parrot Disco.

@szorba the uninstaller script has to be used to clear the zerotier network configuration. Changing the conf file via telnet/ftp will not fix it if you want to change it. Please run the uninstaller script and post screenshots if there are any errors.

SoftIce