uavpal/disco4g #35: telnet now working

buckyuk, 2018-08-15:
Hi guys,
Im at the stage where im trying to enter the commands in telnet, every telnet provider i try to use, it seems to connect, but will not let me type any commands.
I am on a mac, tried using ssh on terminal, is that the same?
Tried using multiple telnet apps on iphone, same problem with it seemingly connecting, but not allowing any commands.
i am using 192.168.42.1 port 21, is this correct?
Thanks so much for any advice

uavpal, 2018-08-15:
just type
`telnet 192.168.42.1`
on your Mac in the Terminal.

Suggest you join the community at https://uavpal.com/slack
Cheers, SoftIce

buckyuk, 2018-08-15:
aparently telnet has been removed from the latest versions of macos, 

that returns command not found

uavpal, 2018-08-15:
try
`nc 192.168.42.1 23`
instead.

Suggest you join the community at https://uavpal.com/slack if you need help during the installation.

buckyuk, 2018-08-15:
it says operation successful, so seems like its connected

but doesnt seem to have a cursor like its ready to accept commands, i paste the command lines from guide, but its not doing anything.

buckyuk, 2018-08-15:
ok i dug out my old windows laptop, and got it working with windows & PuTTY. 
Thanks for your help, although i am now getting an error when trying to install the skycontroller file, its saying trying to connec tto skycontroller 2 via wifi repeatedly.
The light on skycontroller is orange?
I guess i will look through the FAQ's.
Thanks for help
Buckyuk

jugadj, 2018-08-15:
If you don't have homebrew, install it first:
/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"

Then install telnet:
brew install telnet

uavpal, 2018-08-15:
orange LED on SC2 means low battery. charge first! I'll close this issue hereby.

And once more: I suggest you join the community at https://uavpal.com/slack if you need help during the installation!

closed 2018-08-15 by uavpal