uavpal/disco4g #36: Ugly loop may have serious impact on udev on Disco

BadWolf42, 2018-08-26:
Hello,

I went through the Disco4G code and it's an impressive work!
I wondered why the program is kept running when I stumble on the "Uggly loop" : 
https://github.com/uavpal/disco4g/blob/7c1344970a3e0f4bb0db803a85edb29a42b125d4/disco/uavpal/bin/uavpal_disco.sh#L139-L141

As explained here: http://reactivated.net/writing_udev_rules.html
> This program can act on the device, however it **must not run for any extended period of time**, because udev is effectively paused while these programs are running. One workaround for this limitation is to make sure your program immediately detaches itself.

Two options :
- If the program doesn't have to run continuously and is just here to check if it has been run, a lock file, containing the script PID while running and 0 upon successful execution, can be created to avoid a re-run.
- If it is mandatory for the program to run continuously, the best way is to create a daemon and start it in the udev rule. The daemon init script will check if it's already running and the appropriate action to take.

What do you think ?
I have the knowledge to implement both scenario but prefer to ask for your lights and guidance first ;)

Regards,
Bad

uavpal, 2018-08-28:
Thanks BadWolf, the loop can and will be removed in the next release. It is apparently not necessary anymore as all processes initiated by uavpal_disco.sh (scripts and daemons) are detached/non-blocking.
SoftIce

uavpal, 2018-08-28:
https://github.com/uavpal/disco4g/commit/2910e6b68a440d94c83a218e539a91ef8ee9426a

closed 2018-08-28 by uavpal