uavpal/disco4g #37: Problem with E3372h-153 (stick mode)

dipldon, 2018-08-31:
Hello,

i´m siting here for hours to get 4g mod run but it wont work.

I followed installation steps  and already connect SK2 on ZTier.

But  my disco won´t to connect to internet (blue LED flashes once at two sek).

I even downgrade disco´s firmware to 1.7.0 but with same result. 

When i connect my modem on pc with MobilePartner it works fine. Steady blue LED on E3372 and internet.

I don't want to put my modem in HiLink mode, or should i ?

Maybe some changes in /dicso/conf/lte file?

My Provider is T-Mobile Germany, APN: internet.t-mobile  (also tried internet.telekom, nope)

Any Idea?

uavpal, 2018-09-01:
Can you try to change /dev/ttyUSB1 to /dev/ttyUSB0 in disco/uavpal/conf/lte ?

mtony61, 2018-09-01:
salut dipldon , 
j'ai eu le même problème que toi au début 
j'ai voulu le mettre en stick mode , impossible de se connecter disco a ZTIER 
le led bleu flash 

du coup , j'ai mis en hilink , et la , plus de problème , la led est fixe
et ca vole très bien ...

dipldon, 2018-09-01:
salut mtony61,

merci mais je voudraie entendre...(my frensh is terrible, much more then my english)

just tried ot /dev/ttyUSB0     ...... NOPE   still blinking blue

dipldon, 2018-09-01:
yes i restart every time !

now will try to solve on slack...

dipldon, 2018-09-02:
ok, mission softmod can go on :leichtes_lächeln: steady blue with new (same firmware, downloaded on my own, now the e3372 stick pops up as a drive in explorer) newly flashed firmware (E3372h-153_Update_21.315.01.00.143_M_U_01)

closed 2018-09-02 by dipldon