uavpal/disco4g #38: Problem tracking with Glympse

Killianre, 2018-09-02:
Hello,
i have a problem with glympse: i receveid the sms but i only have https://glympse.com/ so i can't track the disco. Like one month i've got the number and the tracking works. A solution ?

uavpal, 2018-09-02:
The link should look like https://glympse.com/abcd-efgh
If that's not there, usually because the Glympse API key is not or wrongly configured in the config file.
SoftIce

Killianre, 2018-09-02:
i have reinstall the Glympse API key but the problem is always there

uavpal, 2018-09-02:
do you mind sharing your glympse_apikey config file? you can send it via email to info@uavpal.com so it's not publicly visible for everyone.

Killianre, 2018-09-02:
ok i send you the file

johanjohn23, 2018-09-02:
Hi I am having the exact same problem! Can you please help me out with this as well?

Thanks,
Johan

uavpal, 2018-09-03:
Yes, everyone was affected. It seems there was a server issue on Glympse's side. I received one report that it's working again. Can you try and let me know? I can't test it myself right now. Will open a ticket with Glympse.
Thanks

pwrkiter, 2018-09-03:
Hi guys, same problem here. It was working last time I tried (15/08/2018) but today when I receive the pushbullet notification it is missing the usual 6 or 8 character link after the glympse.com and I only get:

DISCO-XXXXXX's GPS location: You can track the location of your DISCO-XXXXXX here: https://glympse.com/

Not sure if it's related but I had to re-enable the "Allow ethernet bridging" checkboxes on both devices on ZeroTier today as they had unchecked themselves.

pwrkiter, 2018-09-03:
Just seen your latest post, I'll give it a try in the morning. Thanks!

uavpal, 2018-09-03:
no, not related to zerotier at all...

uavpal, 2018-09-03:
It's working again. It was a Glympse server issue, nothing much we could do about it apart from contacting their support. I'll close the ticket hereby.
SoftIce

closed 2018-09-03 by uavpal

johanjohn23, 2018-09-03:
Thanks! Yeah it’s working for me again.

skyer29, 2021-11-09:
Hi guys, 
I' m having the same problem.
The new glympse .sh file in BIN folder doesn't work.
How did you resolved this problem?

s0ftice, 2021-11-10:
that was a different issue. what you are looking for is here: https://github.com/uavpal/disco4g/issues/88