uavpal/disco4g #41: Script will not run in cmd

mannyscrivener, 2018-10-12:
Hi every, im trying to install the soft mod and im getting as far as running the install command copied and pasted straight from the instruction page and all im getting is this, 

/ # my /data/ftp/internal_000/disco4g-*  /tmp/disco4g/tmp/disco4g/*/*_install.sh
/tmp/disco4g/disco/discoinstall.sh
/tmp/disco4g/skycroller2/skycroller2_install.sh
Rebootmv:  can't overwrite non directory with directory
/  # chmod +×  /tmp/disco4g/*/_install.sh
chmod:  /tmp/disco4g/*/_install.sh:  Not a directory
/  #  /tmp/disco4g/disco/disco_install.sh
/bin/sh:  /tmp/disco4g/disco/disco_install.sh:  not found
/  #  /tmp/disco4g/skycontroller2/skycontroller2_install.sh
/bin/sh:  /tmp/disco4g/skycontroller2/skycontroller2_install.sh: not found
/ # reboot.

I have spent 3 hours going over and over the instructions following them to the T.

Any help would be HUGELY appreciated thank you

uavpal, 2018-10-12:
you typed "my" instead of "mv". I strongly recommend to copy/paste the commands and not to type them manually. Also do line by line and not all at once.

mannyscrivener, 2018-10-12:
Hi, thank you for the response and apologies for the typo. I did copy and paste the commands into cmd. The reason there is a typo there is because I'm using my phone to post this and read it off the screen. I was trying to copy and paste all of it at once so i will try doing it seperately now.. thanks!

uavpal, 2018-10-12:
ok, good luck. I suggest you join us on Slack: https://uavpal.com/slack if you need help.
SoftIce

closed 2018-10-13 by uavpal