uavpal/disco4g #42: Flashing magenta light!

Reyy15, 2018-10-16:
Hi guys! Im a drone pilot from Uruguay (south america) sorry for my english

I do the softmod this afternoon, everything seems to work great, my disco appears on ZeroTier network but i cant figure how to connect the skycontroller to my iPhone X.

I put the correct ssid and password, but when i double press the config button, the skycontroller just flashes magenta forever, i tried another name in the ssid, and re do all the mod to change the network name on the skycontroller, i tried with other cellphone (a samsung), the skycontroller wont connect to the hotspot (the iphone)

i dont know what more i can do, i can do a video for you guys if you want!

uavpal, 2018-10-16:
Do you have any apostrophes in your SSID?
I suggest you join us on Slack: https://uavpal.com/slack

uavpal, 2018-10-19:
Issue solved!
Quick summary: Uruguay as a country is not available in FFpro to choose from as "Wifi-relevant country". @Rey15 had it set to Argentina in FFpro instead. We asked him to change it to US, as we suspected it could be a Wifi channel issue. He tried that, however, it did not solve the issue.

A check on SC2 (via adb) revealed:
```
# wifid-cli status
mode      : STA
country   : name:'UY', selection mode:Auto, regulation known:1, outdoor_mode:Outdoor
channel   : band:unexpected band, channel:-1, selection band:2.4Ghz|5.0Ghz, selection mode:Auto 
```
The country change in FFpro did not get propagated to SC2 for some reason. The "Wifi" country was set to Uruguay (UY), even though @Reyy15 could not able to select it in FFpro. In addition, the band showed "unexpected band", something we have not seen before.

A simple `wifid-cli set_country US` fixed it (for good).

SoftIce

closed 2018-10-19 by uavpal