uavpal/disco4g #46: Help

Skullz81, 2018-12-21:
I’ve done everything listed and turned it all on but when I hit take off nothing happens and the discos light flash’s blue but everything says online. I have video feed.

s0ftice, 2018-12-21:
that's not a softmod issue, that's the general behaviour: Disco LED flashing blue = not strong enough GPS signal (usually indoors), so the motor won't start - same on a stock Wi-Fi connection.

closed 2018-12-21 by s0ftice

Skullz81, 2018-12-21:
No problems, I didn’t know where to post. Was just asking for help. Guess I’ll delete it all out and fly wifi

s0ftice, 2018-12-21:
You misunderstood. It also won't fly Wifi if the GPS signal is not strong enough. Please read my answer again

Skullz81, 2018-12-21:
Flys wifi just fine. Even if I’m in the house testing it. Just 4G mod - won’t take off when told.

s0ftice, 2018-12-21:
step outside and try it again please... maybe you placed the 4G modem right on top of the GPS sensor and that's why it has a harder time acquiring a clear GPS signal from the satellites when you're on 4G?