uavpal/disco4g #50: (Home) wifi interference

fpv88, 2019-01-10:
I’ve installed the 4g mod in my disco, it seems to work, but having trouble connecting my skycontroler 2 over wifi. As soon as there is another wifi accespoint (home router) the connection drops and connecting (green light) is no longer possible.

s0ftice, 2019-01-11:
interesting. is your home router set to the same SSID as your mobile hotspot? do you have the skycontroller 2 with black or silver sticks?

fpv88, 2019-01-11:
Thanks for your quick reply! The ssid was diiferent and it is the silver stick version.

I have down and upgraded the firmware of both and reinstalled mod 1.2.  Everything seems to work perfect now :-) !!

Going to fix the hardware and prepare for maiden flight today.  Great work uavpal!!  Thanks!

s0ftice, 2019-01-11:
Awesome! Join our community on https://uavpal.com/slack great discussion among 4G pilots on the softmod, battery mods and much more!
SoftIce

closed 2019-01-11 by s0ftice