uavpal/disco4g #51: 4g modem doesn't connect

eidos85, 2019-01-10:
Hi,
I've just installed the softmod but after a restart i can't see the disco joining on the ZeroTier network, executing
ulogcat | grep uavpal
I always get this:
```
I uavpal_connmgr(ulogger)                      : ... modem disconnected, trying to reconnect
I uavpal_connmgr(ulogger)                      : ... establishing connection to mobile network
I uavpal_connmgr(ulogger)                      : ... setting IP (10.135.203.234), netmask (255.255.255.252), default gateway (10.135.203.233) and DNS (62.13.169.94, 62.13.169.95)
I uavpal_connmgr(ulogger)                      : ... modem disconnected, trying to reconnect
I uavpal_connmgr(ulogger)                      : ... establishing connection to mobile network
I uavpal_connmgr(ulogger)                      : ... setting IP (10.146.64.20), netmask (255.255.255.248), default gateway (10.146.64.17) and DNS (62.13.173.94, 62.13.173.95)
I uavpal_connmgr(ulogger)                      : ... modem disconnected, trying to reconnect
I uavpal_connmgr(ulogger)                      : ... establishing connection to mobile network
I uavpal_connmgr(ulogger)                      : ... setting IP (10.167.119.105), netmask (255.255.255.252), default gateway (10.167.119.106) and DNS (62.13.173.94, 62.13.173.95)
I uavpal_connmgr(ulogger)                      : ... modem disconnected, trying to reconnect
I uavpal_connmgr(ulogger)                      : ... establishing connection to mobile network
I uavpal_connmgr(ulogger)                      : ... setting IP (10.146.74.217), netmask (255.255.255.252), default gateway (10.146.74.218) and DNS (62.13.173.94, 62.13.173.95)
```
repeating over and over again, the first part of the log is here:
```
I uavpal_drone(ulogger)                        : Huawei USB device disconnected
I uavpal_drone(ulogger)                        : ... unloading scripts and daemons
I uavpal_drone(ulogger)                        : ... removing lock files
I uavpal_drone(ulogger)                        : ... unloading kernel modules
I uavpal_drone(ulogger)                        : *** idle on Wi-Fi ***
I uavpal_drone(ulogger)                        : Huawei USB device detected (USB ID: )
I uavpal_drone(ulogger)                        : === Loading uavpal softmod 1.2 ===
I uavpal_drone(ulogger)                        : ... detected Parrot Disco (platform evinrude), firmware version 1.7.1
I uavpal_drone(ulogger)                        : ... trying to use kernel modules compiled for firmware 1.7.0
I uavpal_drone(ulogger)                        : ... loading tunnel kernel module (for zerotier)
I uavpal_drone(ulogger)                        : ... loading USB modem kernel modules
I uavpal_drone(ulogger)                        : ... loading iptables kernel modules (required for security)
I uavpal_drone(ulogger)                        : ... applying iptables security rules
I uavpal_drone(ulogger)                        : Huawei USB device detected (USB ID: 12d1:1f01)
I uavpal_drone(ulogger)                        : ... running usb_modeswitch to switch Huawei modem into ncm mode
I uavpal_drone(ulogger)                        : === Loading uavpal softmod 1.2 ===
I uavpal_drone(ulogger)                        : ... detected Parrot Disco (platform evinrude), firmware version 1.7.1
I uavpal_drone(ulogger)                        : ... trying to use kernel modules compiled for firmware 1.7.0
I uavpal_drone(ulogger)                        : ... loading tunnel kernel module (for zerotier)
I uavpal_drone(ulogger)                        : ... loading USB modem kernel modules
I uavpal_drone(ulogger)                        : ... loading iptables kernel modules (required for security)
I uavpal_drone(ulogger)                        : ... applying iptables security rules
I uavpal_drone(ulogger)                        : ... detected Huawei USB modem in ncm mode (USB ID: 12d1:155e)
I uavpal_drone(ulogger)                        : ... starting connection manager script
I uavpal_drone(ulogger)                        : ... waiting for public Internet connection
I uavpal_connmgr(ulogger)                      : ... establishing connection to mobile network
I uavpal_drone(ulogger)                        : ... running usb_modeswitch to switch Huawei modem into ncm mode
I uavpal_drone(ulogger)                        : ... detected Huawei USB modem in ncm mode (USB ID: 12d1:155e)
I uavpal_drone(ulogger)                        : ... starting connection manager script
I uavpal_drone(ulogger)                        : ... waiting for public Internet connection
I uavpal_connmgr(ulogger)                      : ... establishing connection to mobile network
I uavpal_connmgr(ulogger)                      : ... setting IP (10.167.44.26), netmask (255.255.255.252), default gateway (10.167.44.25) and DNS (62.13.173.94, 62.13.173.95)
I uavpal_connmgr(ulogger)                      : ... setting IP (10.167.44.26), netmask (255.255.255.252), default gateway (10.167.44.25) and DNS (62.13.173.94, 62.13.173.95)
I uavpal_connmgr(ulogger)                      : ... modem disconnected, trying to reconnect
```
The same modem connected to a pc with the same sim card works great and I've double checked APN and all the other configuration without figuring out what is wrong.
The modem is a e3372h-153.
Any help is appreciated.
Thank you

s0ftice, 2019-01-11:
Hi,
I this a Hi-Link modem? (does a webbrowser open automatically when you plug it into your PC?)
Can you try version 1.1 of the softmod and see if that works? Older releases of the softmod can be found here: https://uavpal.com/disco/faq#releasenotes

Feel free to join us on Slack for more real-time conversations (and meeting  great community of 4G pilots): https://uavpal.com/slack

SoftIce

eidos85, 2019-01-11:
It’s Hi-link version, in order to install a different version I have to uninstall the current one? If yes how can I do?
Thank you

eidos85, 2019-01-11:
Just tried the 1.1 and it worked on the first try (I still have to configure the sky controller and test everything but at least the parrot shows up in the ZeroTier clients)
Thank you

s0ftice, 2019-01-11:
Ok great. There is a small population of modems that don’t work properly on v1.2 anymore. I am still collecting information to try to narrow it down and hopefully fix it in a later release. Do you mind sharing more details on the modem (firmware version) and carrier/apn you are using? I saw you joined our Slack workspace, you can also send me a private message there if you prefer.

Thanks,
SoftIce

ToddTsao, 2019-01-12:
I suffer the same fail on V1.2 & 1.3B today.

s0ftice, 2019-01-13:
Hey @ToddTsao , ignore v1.3-beta, as per today, it's 100% identical with v1.2, only difference is version.txt.
I remember you had two modems (S and H) - do you have the issue with v1.2 on both modems?
I'm trying to find out whether it's modem firmware, hardware or carrier-related. Can you send me as much information as possible on the modem that does not work? (Model, hardware version, firmware version, `ulogcat |grep uavpal`).

Ideally on Slack, I saw you just joined.

Thanks,
SoftIce

ToddTsao, 2019-01-14:
> Hey @ToddTsao , ignore v1.3-beta, as per today, it's 100% identical with v1.2, only difference is version.txt.
> I remember you had two modems (S and H) - do you have the issue with v1.2 on both [...]

Currently, I only have to try H type on V1.2. I will to try S type again.

s0ftice, 2019-01-23:
Upgrading the firmware to 22.328.62.00.143 for E3372**h**-153 and E3372**h**-607 modems fixed the issue.
Download link: https://slack-files.com/T6V4WS6A3-FFLAFJN0N-e73e4910ac

closed 2019-01-23 by s0ftice