uavpal/disco4g #53: "Error while copying files to Skycontroller 2, trying again"

hsy77101, 2019-01-19:
I tried to install by WiKi. But I met problem soon.
The followings are some of my problem. 
Please let me know how to solve them

1st. the following directory is not same with real directory. (Solved)
/tmp/disco4g/disco/disco_install.sh -> "/tmp/disco4g/disco4g-master/disco/disco_install.sh"

2nd. when I run the following commend, I saw them.
/tmp/disco4g/skycontroller2/skycontroller2_install.sh

== installing UAVPAL softmod on Skycontroller 2 ==
Trying ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
* daemon not running. starting it now on port 5037 *
Trying to connect to Skycontroller 2 (192.168.42.74) via adb
not successful, trying again
Successfully connected to Skycontroller 2!
Copying softmod files from drone to Skycontroller 2
Error while copying files to Skycontroller 2, trying again
Error while copying files to Skycontroller 2, trying again
Error while copying files to Skycontroller 2, trying again
Error while copying files to Skycontroller 2, trying again
Error while copying files to Skycontroller 2, trying again
Error while copying files to Skycontroller 2, trying again
Error while copying files to Skycontroller 2, trying again
Error while copying files to Skycontroller 2, trying again
Error while copying files to Skycontroller 2, trying again
Error while copying files to Skycontroller 2, trying again
Error while copying files to Skycontroller 2, trying again
Error while copying files to Skycontroller 2, trying again

s0ftice, 2019-01-20:
Hi there,

The instructions say: _Transfer the "disco4g-master" or "disco4g-x.y" folder to the "/internal_000" folder on the Disco._
It seems you transferred something like `disco4g-master/disco4g-master` to your Disco. Can you send a screenshot of the target folder structure after your FTP transfer?

2.) probably related to 1.) as it can't find the toolchain required in the (wrong) path.

SoftIce

s0ftice, 2019-01-22:
you got it fixed? can I close the issue?

hsy77101, 2019-01-22:
I did't solve this problem. I will try again like mentioned way by you.
And then I will let you know that.
Thank you .

dakinehvac, 2019-01-30:
wondering if anyone could help me with a similar problem. the disco is online on zerotier, skycontroller2 is not and blinking red. At cmd install first three lines installed until /tmp/disco4g/skycontroller2/skycontroller2_install.sh file not found. Not sure what to do now

s0ftice, 2019-01-30:
Please re-do the installation and post a screenshot here if you get an error again. I would need to see the entire telnet session on that screenshot.

Ideally, also post a screenshot of the FTP tool after you transferred the files.

SoftIce

dakinehvac, 2019-01-30:
Status:	Connecting to 192.168.42.1:21...
Status:	Connection established, waiting for welcome message...
Status:	Insecure server, it does not support FTP over TLS.
Status:	Server does not support non-ASCII characters.
Status:	Logged in
Status:	Retrieving directory listing...
Status:	Directory listing of "/" successful
Status:	Connecting to 192.168.42.1:21...
Status:	Connection established, waiting for welcome message...
Status:	Connecting to 192.168.42.1:21...
Status:	Connection established, waiting for welcome message...
Status:	Insecure server, it does not support FTP over TLS.
Status:	Server does not support non-ASCII characters.
Status:	Logged in
Status:	Starting upload of C:\Users\Keith2\Documents\disco4g-1.2\skycontroller2\skycontroller2_uninstall.sh
Status:	Insecure server, it does not support FTP over TLS.
Status:	Server does not support non-ASCII characters.
Status:	Logged in
Status:	Starting upload of C:\Users\Keith2\Documents\disco4g-1.2\skycontroller2\skycontroller2_install.sh
Status:	Retrieving directory listing of "/internal_000/skycontroller2"...
Status:	Retrieving directory listing of "/internal_000/skycontroller2"...
Status:	File transfer successful, transferred 1,908 bytes in 1 second
Status:	Starting upload of C:\Users\Keith2\Documents\disco4g-1.2\skycontroller2\uavpal\version.txt
Status:	File transfer successful, transferred 1,258 bytes in 1 second
Status:	Starting upload of C:\Users\Keith2\Documents\disco4g-1.2\skycontroller2\uavpal\bin\uavpal_sc2.sh
Status:	Retrieving directory listing of "/internal_000/skycontroller2/uavpal/bin"...
Status:	Retrieving directory listing of "/internal_000/skycontroller2/uavpal"...
Status:	File transfer successful, transferred 4 bytes in 1 second
Status:	Starting upload of C:\Users\Keith2\Documents\disco4g-1.2\skycontroller2\uavpal\bin\zerotier-one
Status:	File transfer successful, transferred 13,206 bytes in 1 second
Status:	Starting upload of C:\Users\Keith2\Documents\disco4g-1.2\skycontroller2\uavpal\conf\local.conf
Status:	Retrieving directory listing of "/internal_000/skycontroller2/uavpal/conf"...
Status:	File transfer successful, transferred 75 bytes in 1 second
Status:	Starting upload of C:\Users\Keith2\Documents\disco4g-1.2\skycontroller2\uavpal\conf\ssid
Status:	File transfer successful, transferred 166 bytes in 1 second
Status:	Starting upload of C:\Users\Keith2\Documents\disco4g-1.2\skycontroller2\uavpal\conf\wpa
Status:	File transfer successful, transferred 167 bytes in 1 second
Status:	Starting upload of C:\Users\Keith2\Documents\disco4g-1.2\skycontroller2\uavpal\conf\zt_networkid
Status:	File transfer successful, transferred 153 bytes in 1 second
Status:	File transfer successful, transferred 1,349,188 bytes in 1 second
Status:	Disconnected from server
Status:	Disconnected from server

dakinehvac, 2019-01-30:
after I did this tonight I realized the sky controller was off

dakinehvac, 2019-01-30:
[image]
wondering if this helps

s0ftice, 2019-01-30:
You posted “after I did this tonight I realized the sky controller was off”, but you deleted the message I think. If that’s the case, you need to start the installation ffrom scratch anyways.

The screenshot you posted is just the one step. You need to start from scratch.

Looking at the ftp log your target directory is wrong, please check the instructions again.

s0ftice, 2019-02-02:
guys, I'm closing this issue. If you need help, head over to https://uavpal.com/slack

closed 2019-02-02 by s0ftice