uavpal/disco4g #56: Can't connect to Disco via ftp 51 (port closed)...

Murgator, 2019-02-17:
Hi ! I tried to install the 4G mod in my Parrot Disco 1.7.1 that I bought in January 2019 but I can not connect via port 51 or 21 with filezilla.

1) I establish a wifi connection between my PC and my Disco (Disco IP address is: 192.168.42.44), the connection is a success.
2) Then I try an ftp connection with filezilla at 192.168.42.44:51, but the connection is refused (closed port).

So I did a port scan with nmap and here's what I got:

[image]

Do you have a solution ?
Thank you in advance.
David

s0ftice, 2019-02-17:
192.168.42.44 is probably your own (Windows PC’s) IP address. The Disco is always 192.168.42.1.
Just follow the instructions one by one, can’t really go wrong.

SoftIce

Murgator, 2019-02-17:
Thank you for your response Softice the problem has fix, I simply start the process again...

But i have another problem with my 4h modem. I open a new ticket.

closed 2019-02-18 by s0ftice