uavpal/disco4g #57: My E3372s - 153 4G Modem can't connect, zerotier member not created.

Murgator, 2019-02-17:
Hi ! I followed all the installation steps of 4G 1.2 mod successfully and when I typed REBOOT in TELNET console the disco restarted.

The Huawei E3372s - 153 does not connect (flashing green led for several minutes).

So I plugged the key into my computer and disabled pin code protection and enable automatic login.

I manually restarted the disco with the Huawei E3372s.

The led of the Huawei E3372s has become blue fixed, so it is connected to the internet but the disco does not connect to zerotier (no member is created).

Here is the firmware version of the Huawei E3372s 4G key:

[image]

An idea ?
Thank you in advance
David.

s0ftice, 2019-02-18:
Can you send us the output of the log on the disco? See details here: https://uavpal.com/disco/faq#discodebug

I suggest you join us on Slack: https;//uavpal.com/slack
You'll have an entire community to help and troubleshoot your setup :)

Thanks,
SoftIce

Murgator, 2019-02-18:
here is my logs:

[image]

I tested with 1.1 mod and it's work, but not with 1.2....

Ok thank you i will join the community

krwi, 2019-02-25:
Same here, repeated modem disconnects, no ZT connection, no Glympse notification.
Hilink E3372, worked like a charm in disco4g v1.0.

Murgator, 2019-02-25:
Fixed by changing modem to a e3372h 153 and create another zerotier networf

closed 2019-02-25 by Murgator

JeanVautrin, 2019-03-03:
Hi Murgator
I am considering the Disco4G softmod. I received the E3372h - 153.
Now I hesitate choosing the mobile 4G provider.
I plan to get 2 40Go/50Go sim cards sans engagement.
Would you advise RED/SFR or Sosh/Orange?
Cordially

s0ftice, 2019-03-03:
@JeanVautrin I recommend you join us on Slack: https://uavpal.com/slack, you can reach out to our French members there.
SoftIce

Murgator, 2019-03-03:
JeanVautrin j'ai pris cet abonnement SFR 4G à 3€ par jour d'utilisation car je ne voulais pas payer un abonnement mensuel: https://www.sfr.fr/box-internet/internet-partout.html

JeanVautrin, 2019-03-03:
Merci Murgator

Merci pour ta réponse. Très bonne info mais SFR semble très mal couvrir 4G mon village et j’ai peur d’avoir du mal à faire les mises au point.
Comme je l’ai dit sur UAVPAL (alias jyvai) j’envisage Sosh Orange.

A+

1019144701, 2019-03-15:
E3372s - 153 and E3372h - 607 can't connect , now I will buy  a E3372h - 153 for a try.