uavpal/disco4g #58: Using ZeroTier over 4G LTE for in flight drone video (snowy picture)

mattgorski, 2019-03-07:
I came across your project and I am using a similar setup with QGroundControl with zero tier.  Here is my issue and im hoping you can help me with this.

I am using ZeroTier to provide in flight video/telemetry over 4G LTE for long range flights on my drone.

I tested connection over my ISP using ethernet from my drone to work PC and everything works as expected over ethernet and I love the results, very clear picture.

When using 4G LTE the results are very different. The video picture is snowy and I can not figure out why this is happening. I have IPV4 in my apn settings and my speeds are just as good as my ethernet speeds, about 50mb down and 15mb up (even when connected to ZT). My drone is only broadcasting at 750 kb/s so I know the speeds are optimal on 4G LTE. Am I missing a crucial setting to make this work over cellular? Do I need to route all traffic through ZT, open UDP port 9993 and also create a bridge on my drone running linux for better results?

All clients are using up to date zerotier v 1.2.12.

Drone is running on Ubuntu 16.04 ROS Kinetic with Sierra Wireless mini PCIe LTE onboard connected to Google Fi on wwan0.

Ground Station is an Android Tablet with Google Fi LTE and full signal. Video over wifi on android with ZT works fine 4G LTE is the only issue for picture and packet loss (picture remains snow on 4g lte).

Any help is greatly appreciated.

s0ftice, 2019-03-08:
Hey Matt, please visit our community workspace on Slack. I might have an idea why you don't get the bandwidth and latency required.

I'm closing this issue as it's not a softmod issue.

SoftIce

closed 2019-03-08 by s0ftice

mattgorski, 2019-03-08:
@s0ftice no problem, I will join the slack channel and ask there.  Thank you kindly for your help.