uavpal/disco4g #61: SD Card recording issue

Zap123456789, 2019-06-09:
Hello,

I'm trying the new function "SD" recording, from version 1.3 mod.
I've a SD card FAT32 (allocation unit 32ko)  inside the modem 3372s however when I connect / diconnect the modem, no light are flashing on the disco ... And when I record Video / Pictures, everything is going on the disco and not on the SD Card
I tried to format the SD Card but not working neither.
When I connect to the disco by Filezilla, I can see and access to the SD Card without issue (but there is nothing on it)
I'm using Disco version 1.4.1 and Skycontroller version 1.0.7

s0ftice, 2019-06-09:
Do you receive any push notifications.SMS for Glympse? If so, do you also get something which is related to the SD card? (You should)

s0ftice, 2019-06-09:
Otherwise, please post the ulogcat output here: https://uavpal.com/disco/faq#debug

Zap123456789, 2019-06-09:
I'm receiving from Pushbullet the link to see the Disco location in Glympse notification, but nothing related to SD
I tried to "debug" by doing ulogcat | grep uavpal it's writting :

/ # ulogcat | grep uavpal
I uavpal_glympse(ulogger)                      : ... updating Glympse label (201
9-06-09-08:38:56): DiscoZ (Sig:4G/75% Alt:0m Bat:4%/11.22V Ltn:n/a)

s0ftice, 2019-06-09:
there should be more lines than just this one... I need all of them

Zap123456789, 2019-06-09:
Nothing is coming, I'm waiting for 30 min now...
When I turned of the disco I just receive this message above.
So I tried the other way with the repertory "Debug" 

> : Machine: Evinrude board
K I KERNEL                                       : P7 per device DMA memory regions:
K I KERNEL                                       : 	9ff00000:9fffffff: ramoops.0
K I KERNEL                                       : 	9fe00000:9fefffff: cast-nand.0
K I KERNEL                                       : 	9fd00000:9fdfffff: ci_hdrc.0
K I KERNEL                                       : 	9fc00000:9fcfffff: ci_hdrc.1
K I KERNEL                                       : 	00000000:0cdfffff: avicam.0
K W KERNEL                                       : Memory policy: ECC disabled, Data cache writealloc
K D KERNEL                                       : On node 0 totalpages: 77312
K D KERNEL                                       : free_area_init_node: node 0, pgdat c06c11c0, node_mem_map c076c000
K D KERNEL                                       :   Normal zone: 1024 pages used for memmap
K D KERNEL                                       :   Normal zone: 0 pages reserved
K D KERNEL                                       :   Normal zone: 76288 pages, LIFO batch:15
K I KERNEL                                       : PERCPU: Embedded 8 pages/cpu @c0b71000 s10304 r8192 d14272 u32768
K D KERNEL                                       : pcpu-alloc: s10304 r8192 d14272 u32768 alloc=8*4096
K D KERNEL                                       : pcpu-alloc: [0] 0 [0] 1 
K W KERNEL                                       : Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 76288
K N KERNEL                                       : Kernel command line:  boxinit.serialno=P742P35C750000B19200  calib=0x001C0039 mtdparts=nand0:8M(Pbootloader),16M(Pmain_boot),8M(Pfactory) console=ttyPA0,115200 loglevel=8 ubi.mtd=Pfactory root=/dev/mmcblk0p3 rootfstype=ext4 rootwait   androidboot.bootloader=rtv-ecos-bootloader-p7-0-38-12-g11f6a6e 
K I KERNEL                                       : PID hash table entries: 2048 (order: 1, 8192 bytes)
K I KERNEL                                       : Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
K I KERNEL                                       : Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
K I KERNEL                                       : Memory: 302MB = 302MB total
K N KERNEL                                       : Memory: 297016k/297016k available, 227272k reserved, 0K highmem
K N KERNEL                                       : Virtual kernel memory layout:
K N KERNEL                                       :     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
K N KERNEL                                       :     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
K N KERNEL                                       :     vmalloc : 0xe0800000 - 0xff000000   ( 488 MB)
K N KERNEL                                       :     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
K N KERNEL                                       :     modules : 0xbf000000 - 0xc0000000   (  16 MB)
K N KERNEL                                       :       .text : 0xc0008000 - 0xc06428c8   (6379 kB)
K N KERNEL                                       :       .init : 0xc0643000 - 0xc066e840   ( 175 kB)
K N KERNEL                                       :       .data : 0xc0670000 - 0xc06c36b0   ( 334 kB)
K N KERNEL                                       :        .bss : 0xc06c36d4 - 0xc076ba90   ( 673 kB)
K I KERNEL                                       : Preemptible hierarchical RCU implementation.
K I KERNEL                                       : 	Dump stacks of tasks blocking RCU-preempt GP.
K I KERNEL                                       : NR_IRQS:202
K D KERNEL                                       : p7: IRQ setup
K D KERNEL                                       : p7: found Parrot7 chip v3 [0x50373032]
K D KERNEL                                       : p7: tick setup
K I KERNEL                                       : sched_clock: 32 bits at 390MHz, resolution 2ns, wraps every 11012ms
K I KERNEL                                       : Console: colour dummy device 80x30
K I KERNEL                                       : Calibrating delay using timer specific routine.. 1561.64 BogoMIPS (lpj=7808216)
K I KERNEL                                       : pid_max: default: 32768 minimum: 301
K I KERNEL                                       : Mount-cache hash table entries: 512
K I KERNEL                                       : CPU: Testing write buffer coherency: ok
K I KERNEL                                       : CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
K I KERNEL                                       : hw perfevents: enabled with ARMv7 Cortex-A9 PMU driver, 7 counters available
K I KERNEL                                       : Setting up static identity map for 0x80488558 - 0x8048858c
K D KERNEL                                       : p7: powering built-in self test off
K D KERNEL                                       : p7: setting L2C latencies to tag=00000020 data=00000020.
K I KERNEL                                       : L310 cache controller enabled
K I KERNEL                                       : l2x0: 16 ways, CACHE_ID 0x410004c5, AUX_CTRL 0x72450801, Cache size: 524288 B
K W KERNEL                                       : CPU1: Booted secondary processor
[... cut ...]
U E MassStorage (dragon-prog/MassStorage)        : Error while polling for events
U E MassStorage (dragon-prog/MassStorage)        : Error while polling for events
U E MassStorage (dragon-prog/MassStorage)        : Error while polling for events
U E MassStorage (dragon-prog/MassStorage)        : Error while polling for events
U E MassStorage (dragon-prog/MassStorage)        : Error while polling for events
U E MassStorage (dragon-prog/MassStorage)        : Error while polling for events

>

And this message again and again, like 1000 times

s0ftice, 2019-06-09:
can you upload the entire debug log file, there should be more in there...

Zap123456789, 2019-06-09:
Sure, here we go
[ulog_debug_20190609092417.log](https://github.com/uavpal/disco4g/files/3269608/ulog_debug_20190609092417.log)

s0ftice, 2019-06-09:
nah it only starts at
`I uavpal_drone(ulogger)                        : ... Debug mode is enabled - writing debug log to internal storage: /data/ftp/internal_000/Debug/ulog_debug_20190609092417.log`

I need to see the earlier events...
It's really weird, as `ulogcat |grep uavpal` should give you all we need. I.e. when running it after you received the Push Notification. Can you try that?

Zap123456789, 2019-06-09:
File is getting bigger and bigger with time, at the end 278 Mo this morning when I was waiting 30 min to have something display with the telnet. More I wait, heavier it is.
See attached last file, attached file already 25 Mo after some minutes and notification received
=> I believe this is why ulogcat | grep uavpal is loading again and again without displaying anything, message should be too big.

[ulog_debug_20190609144149.log](https://github.com/uavpal/disco4g/files/3269646/ulog_debug_20190609144149.log)

s0ftice, 2019-06-10:
No, not the debug log, as it's missing out the lines before it gets enabled (which is actually weird too).
As said, run `ulogcat |grep uavpal`  after you received the Glympse Push Notification.

Zap123456789, 2019-06-10:
This is what I did, but I've nothing coming out

s0ftice, 2019-06-10:
Something is completely off :)
Yes, the Glympse link via Push Bullet. Can you try to

1. power on the Disco with the modem+SIM attached, but **without** SD card inside
2. once you receive the Glympse/Push Bullet notification, run `ulogcat |grep uavpal` (leave it running). This should already give you more output than before. Upload that here.
3. slide in the SD card. Does that trigger a second notification? What does the running `ulogcat |grep uavpal` show now in addition to the previous lines?

It could be that the error "U E MassStorage (dragon-prog/MassStorage) : Error while polling for event" occurs so many times, that the earlier ulog buffer gets removed.

If nothing works, can you upgrade the Disco to 1.7.1 and try again? (you can always downgrade to 1.4.1). However the softmod needs to be reinstalled after up/downgrading.

Zap123456789, 2019-06-10:
2. Without SD Card :

> BusyBox v1.25.0 (2017-05-31 10:55:10 CEST) built-in shell (ash)
> Enter 'help' for a list of built-in commands.
> 
> / # ulogcat | grep uavpal
> I uavpal_drone(ulogger)                        : Huawei USB device detected (USB
>  ID: 12d1:14fe)
> I uavpal_drone(ulogger)                        : === Loading uavpal softmod 1.3
> ===
> I uavpal_drone(ulogger)                        : ... detected Parrot Disco (plat
> form evinrude), firmware version 1.4.1
> I uavpal_drone(ulogger)                        : ... trying to use kernel module
> s compiled for firmware 1.4.1
> I uavpal_drone(ulogger)                        : ... loading tunnel kernel modul
> e (for zerotier)
> I uavpal_drone(ulogger)                        : ... loading USB modem kernel mo
> dules
> I uavpal_drone(ulogger)                        : ... loading iptables kernel mod
> ules (required for security)
> I uavpal_drone(ulogger)                        : ... running usb_modeswitch to s
> witch Huawei modem into huawei-new-mode
> I uavpal_drone(ulogger)                        : ... detecting Huawei modem type
> I uavpal_drone(ulogger)                        : ... detected Huawei USB modem i
> n Stick mode
> I uavpal_drone(ulogger)                        : ... loading ppp kernel modules
> I uavpal_drone(ulogger)                        : ... connecting modem to Interne
> t (ppp)
> I uavpal_connect_stick(ulogger)                : ... running pppd to establish c
> onnection to mobile network using APN "orange.fr" (configured in the conf/apn fi
> le)
> I uavpal_connect_stick(ulogger)                : ... interface "ppp0" is up
> I uavpal_drone(ulogger)                        : ... querying Huawei device deta
> ils via AT command
> I uavpal_drone(ulogger)                        : ... model: E3372S-153, hardware
>  version: CL1E3372SM Ver.A
> I uavpal_drone(ulogger)                        : ... software version: 21.297.05
> .00.00
> I uavpal_drone(ulogger)                        : ... applying iptables security
> rules for interface ppp0
> I uavpal_drone(ulogger)                        : ... starting connection keep-al
> ive handler in background
> I uavpal_drone(ulogger)                        : ... public Internet connection
> is up
> I uavpal_drone(ulogger)                        : ... setting DNS servers statica
> lly (Google Public DNS)
> I uavpal_drone(ulogger)                        : ... setting date/time using ntp
> I uavpal_drone(ulogger)                        : ... starting Glympse script for
>  GPS tracking
> I uavpal_drone(ulogger)                        : ... starting zerotier daemon
> I uavpal_glympse(ulogger)                      : ... reading Glympse API key fro
> m config file
> I uavpal_drone(ulogger)                        : *** idle on LTE ***
> I uavpal_glympse(ulogger)                      : ... reading drone ID from avahi
> I uavpal_glympse(ulogger)                      : ... Glympse API: creating accou
> nt
> I uavpal_glympse(ulogger)                      : ... Glympse API: logging in
> I uavpal_glympse(ulogger)                      : ... Glympse API: parsing access
>  token
> I uavpal_glympse(ulogger)                      : ... Glympse API: creating ticke
> t
> I uavpal_glympse(ulogger)                      : ... Glympse API: parsing ticket
> I uavpal_glympse(ulogger)                      : ... Glympse API: creating invit
> e
> I uavpal_glympse(ulogger)                      : ... Glympse link generated: htt
> ps://glympse.com/0F6G-BMWY
> I uavpal_glympse(ulogger)                      : ... Glympse API: setting drone
> thumbnail image
> I uavpal_send_message(ulogger)                 : ... sending SMS to +33673421879
>  (via ttyUSB0)
[... 143 lines cut ...]
9-06-10-19:07:50): DiscoZ (Sig:4G/66% Alt:0m Bat:38%/11.37V Ltn:n/a)
I uavpal_glympse(ulogger)                      : ... updating Glympse label (201
9-06-10-19:07:57): DiscoZ (Sig:4G/66% Alt:0m Bat:37%/11.36V Ltn:n/a)
I uavpal_glympse(ulogger)                      : ... updating Glympse label (201
9-06-10-19:08:05): DiscoZ (Sig:4G/72% Alt:0m Bat:36%/11.40V Ltn:n/a)
I uavpal_glympse(ulogger)                      : ... updating Glympse label (201
9-06-10-19:08:13): DiscoZ (Sig:4G/63% Alt:0m Bat:36%/11.35V Ltn:n/a)
I uavpal_glympse(ulogger)                      : ... updating Glympse label (201
9-06-10-19:08:21): DiscoZ (Sig:4G/72% Alt:0m Bat:36%/11.35V Ltn:n/a)
I uavpal_glympse(ulogger)                      : ... updating Glympse label (201
9-06-10-19:08:29): DiscoZ (Sig:4G/63% Alt:0m Bat:36%/11.40V Ltn:n/a)
I uavpal_glympse(ulogger)                      : ... updating Glympse label (201
9-06-10-19:08:36): DiscoZ (Sig:4G/66% Alt:0m Bat:35%/11.40V Ltn:n/a)
I uavpal_glympse(ulogger)                      : ... updating Glympse label (201
9-06-10-19:08:44): DiscoZ (Sig:4G/66% Alt:0m Bat:35%/11.35V Ltn:n/a)
I uavpal_glympse(ulogger)                      : ... updating Glympse label (201
9-06-10-19:08:52): DiscoZ (Sig:4G/72% Alt:0m Bat:35%/11.35V Ltn:n/a)
I uavpal_at_command(ulogger)                   : ... Did not receive expected ou
tput from AT command AT+CSQ
I uavpal_glympse(ulogger)                      : ... updating Glympse label (201
9-06-10-19:09:01): DiscoZ (Sig:4G/n/a Alt:0m Bat:35%/11.35V Ltn:n/a)
I uavpal_glympse(ulogger)                      : ... updating Glympse label (201
9-06-10-19:09:09): DiscoZ (Sig:4G/69% Alt:0m Bat:35%/11.35V Ltn:n/a)

```

s0ftice, 2019-06-12:
What happens when you insert **the modem with the SD card** into a PC? Can you access the card's storage on the PC then?

Zap123456789, 2019-06-12:
Yes, and when I insert it into the Disco, I can access to the SD Card via FTP using Filezilla without issue

Do you want me to update 1.7.1 to check if issue is coming from compatibility with version 1.4.1 ?

s0ftice, 2019-06-13:
yes, you can give it a try. Have a look here: https://uavpal.com/disco/faq#firmwareupdate
with this, you won't need to reinstall the softmod after the upgrade.

Zap123456789, 2019-06-14:
I updated to 1.7.1 via normal Parrot update on smartphone
I reinstalled Disco 4G 1.3
Then I tried again to do pictures and it was not going on SD Card neither
I tried : 
ulogcat |grep uavpal 
=> was working fine 

I removed the SD Card from the modem and put it in again => Led flashing with multicolors
Then I tried again to do pictures and it was ... OK :-) :+1: 

Thanks a lot it's now working, so I believe the option is not compatible with 1.4.1, at least from my own experience

Regards

closed 2019-06-14 by Zap123456789

s0ftice, 2019-06-15:
Ok, thanks for the feedback. I will add it to the documentation for now. Maybe I’ll fix it later to make it work on the old version.

Thanks!
SoftIce