uavpal/disco4g #62: Is it possible to control the drone completely autonomous?

zesanriky, 2019-08-08:
I am really excited about this project. I have done autonomous AR drone 2.0 which flies completely autonomous without any remote control from the ground. Can I use this project the same way? 

Can I integrate open CV to it as well?

s0ftice, 2019-08-09:
Not sure if I understand you correctly. Why do you need to combine complete autonomy with a 4G connection? For autonomy, you won't need a connection at all...?

zesanriky, 2019-08-09:
Thanks for the answer. I just want to know is it possible to control it by any script from repository or I must use the joystick to control it manually? I understood it returns back autonomously only.

s0ftice, 2019-08-10:
It can fly autonomously. There is even an integrated Flight Plan feature, where you can design a flight plan on a phone/tablet and send it to the drone. It will continue the plan upon losing connection.
These plans are just simple JSON files with coordinates which can be generated with the right tools as well.

closed 2019-08-10 by s0ftice