uavpal/disco4g #64: Software lock up after temporary signal loss

StevenLawyer, 2019-09-02:
Hello.  Flying a Disco with a large capacity battery mod.  Using the Huawei E3372 modem with a 128Gb microSD for video over Verizon in the US.  Here's the problem I'm facing.  On two separate long range flights so far, I've had the 4g mods "lock up".  Here's what happens:  Somewhere along the flight I hit a spot where the 4g signal is lost (due to distance from an antenna, aircraft orientation, whatever).  Normally, when that happens, when the craft gets back in a good reception area (say in a minute or two), telemetry and FPV comes back and my glympse feed starts updating again.  But twice now, I've had where it never reconnects.  The first time, the Disco was over an hour away.  For an hour I waited watching the skies until it came home.  Something in the Disco had locked up to the point that, even when it got back to me and began loitering, I could not reconnect on the direct wifi in order to land the drone.  I tried controller off and on, etc.  Nothing worked.  After the 5 minute loiter, the Disco auto-landed with me never regaining control.  I powered the Disco off and back on and, as soon as I did so, everything worked including 4g connection and Glympse.  Yesterday, I had a similar experience when I was over an hour and a half into the flight.  I had had two momentary signal losses where they had regained.  On the third one, I never regained telemetry, FPV or glympse.  The drone came all the way home (through the exact same areas it had passed through outbound that have great coverage) with the connection never regaining, with the exception that I got one solitary glympse update about half way back.  This time, I was able to take control via direct wifi when it got back.  Again, after I powered the drone off and back on, everything worked fine.  I have eliminated heat as the likely cause (good ventilation, cool weather, etc).  I am running continual video recording onto the microSD in the Huawaii modem.  On the first flight (the one where even wifi connection didn't work), the video file that was being written at the time of the loss of 4g control was not closed properly--clear that all connection to the modem had stopped.  I don't think it was a cabling issue, since a power off and on immediately restored connection.  On the second flight (where 4g and glympse quit but I could control via wifi), the video continued to record to the SD card.  Any thoughts on what is occurring?

s0ftice, 2019-09-04:
Hi Steven. I replied on Slack to you on Monday already. Guess you did not see it.
There have been occasional reports where a reconnection via 4G is not possible. Can you try to enable the debug logging, this might help to investigate what's going on. Note down the exact time when a disconnection happens the next time.
See https://uavpal.com/disco/faq#debug

Are you on the latest firmware versions of Disco/SC2 and the modem?

I never had this issue myself.

SoftIce

StevenLawyer, 2019-09-07:
S0ftice.  Thanks for replying.  Yes, I'm on the latest firmware on both.  I have enabled debugging.  Should be able to do a flight today.

dru180, 2019-09-16:
Hi guys, has there been any further developments on this issue ? I had exactly the same issue, my disco came home and loitered till the battery was low and did an auto land. I changed the battery, reconnected after 3 power cycles, it all finally connected again. I than tried to commence another flightplan, and an error message on FFP indicated something along the lines of not being able to load the flight plan. 
I'll try running a debug as well.  
Cheers.
Dru.

s0ftice, 2019-09-16:
No update no. Waiting for people to share debug logs.
Just to make sure you understood: The Skycontroller 2 will not fall-back to Wi-Fi automatically. It needs to be put into Wi-Fi mode manually (double-press Settings again). It can also be powered off and on again to reconnect via Wi-Fi.

The problem described here implies that even that did not work, correct?

dru180, 2019-09-16:
Yes, correct, I'll email you the debuts next time I fly. Cheers.

StevenLawyer, 2019-09-16:
So, I had it happen again (I think) on a flight that lasted 4 hours and 5 minutes.  Unfortunately, my battery table was a little off and the drone ended up running out of power and crashing in the woods on the way home.  I did manage to get the debug file off of it and will send it along with the time at which I lost control signal.  I definitely was in good 4G reception.  Unlike the other time that I lost control, I continued to receive my Glympse feed and, when I retrieved the video, it continued to record video (the first time, Glympse stopped, wifi wouldn't reconnect when home, and video recording to SD in Huwaei modem quit at the same moment, leaving a corrupted video file).  On this last occasion,  I can't say for sure whether or not the wifi connection ability locked out.  I tried to drive to rendezvous with the disco on its auto-return and think I got close enough to connect via wifi, but never could get the connection.  However, it's possible that I was not close enough.  I can say that the live video feed locked up even though I got telemetry for a while and then all control locked up, but I still had good 4g connection (according to Glympse feed).  Of course, it's possible that was all do to zerotier problems.

I'll send the times and debug file.

Nicktk23, 2019-09-29:
If the logs are showing that zerotier have issue, would be nice to be added a VPN (client) connection option (conf.file)  to Disco to make connection in same LAN Network with SC2 .

s0ftice, 2019-09-30:
Hi Nick, if a direct connection is possible, zerotier will not relay the connection via their cloud infrastructure. It will just facilitate the initial establishing of connection but all traffic is directly routed between the two devices. It can also do NAT traversal which is required for most carriers due to their internal network design. A normal VPN client does not do that.

Feel free to use your own VPN client, but it will not be as easy as you described it.

Now question to all who reported this issue, do you guys see a [D]irect or [R]elayed connection in Glympse?

StevenLawyer, 2019-09-30:
I had direct.

Steven Lawyer
7129 G-76 Hwy
New Virginia, Iowa 50210
Steven@LawyerFamily.net<mailto:Steven@LawyerFamily.net>
515-229-5352

marph43, 2019-10-05:
On my last 4G-flight my Disco experienced exactely the same error, as originally described by StevenLawyer: When it lost the connection to the 4G network it was not able to reconnect at all, also when the network coverage was good enough again (during flight-to-home). Additionally it was not transmitting anymore to Glympse. 
Even connecting directely to the SC2 by wifi, when it finally showed up at the beginning coordinates did not work anymore (Fortunately it made an automatic circular landing which went well). A reboot of the drone fixed the problem.
I will run a debug on my next flight in case it happens again.

s0ftice, 2019-10-06:
Yes, please post the debug log if you can.
The important question that needs to be clarified is, what is the cause and what is the consequence.

While what you guys describe is a loss of 4G connection causing the Wifi interface to shutdown, might as well be: Loss of all network interfaces that makes 4G disconnect in the first place.

Could it be that the battery used does not provide enough power? Are you all flying on stock battery? The reason why I believe this is not a software issue: No reports of the same on the Bebop2 so far, which uses the same softmod code and modules.

marph43, 2019-10-06:
I will. 

Hmm, interesting point. Indeed, I am using a battery mod as well and it is possible that it can't handle power peaks as well as the stock battery (higher inner resistance). What I do not understand though, how can this only lead to the shutdown of the wirless devices and not of the whole Chuck?

s0ftice, 2019-10-06:
no idea :) but if you guys can find something that all of your setups have in common, we might get a lead. I never had that issue myself.

marph43, 2019-10-09:
Thinking further about this issue, led me to the idea that it could maybe be due to overheating of the Chuck.  This could be the case, because I was flying against rather strong wind for more than 30 minutes. Is there a way to log the core temperature of the Chuck?

StevenLawyer, 2019-10-09:
I’m confident it’s not overheating.  Been very cool the times I’ve had it happen.

Steven Lawyer
7129 G-76 Hwy
New Virginia, Iowa 50210
Steven@LawyerFamily.net<mailto:Steven@LawyerFamily.net>
515-229-5352

marph43, 2019-10-27:
OK. So yesterday it happened again. Somehow. Not sure if the problem was the same or if the network coverage was insufficient. The situation was the follwoing: 

I flew close to a mountain and tried to cross it, but by that time the battery was already half empty. When I stared to fly back home, it suddenly started to loose connection. At first it sort of connected again several times, but bandwith was insufficient to regain control and the video was very very choppy. After some reconnections it disconnected permanently. Because I have an additional GPS tracker installed, I saw that the Disco fought against the wind to return home, but I underestimated the wind at this high altituede and was now unable to pilot the plane to a lower height, so it was not long until the battery was depleated and it stopped returning. Instead it was carried out again by the wind and went down somewhere in the woods. Luckily I was able to retrieve it, because of the additonal GPS tracker.

So now the question is, was it unable to reconnect because of the bad network coverage at this high altitude or because of something else? Like insufficient power? An interesting detail is, that the Chuck was running until the crash (logged everything also the route).

Please have a look at the attached debug log file. I lam looking forward to read your interpretation...

[ulog_debug_20191026150251.log](https://github.com/uavpal/disco4g/files/3775750/ulog_debug_20191026150251.log)

s0ftice, 2019-10-27:
Great, I will look into the log later. One question, when you found the Disco and CHUCK was still running. Did you try to reconnect your SC2 while in Wi-Fi mode? That would give an indication whether the reconnection was not possible at all, or just the 4G reconnection.

StevenLawyer, 2019-10-27:
On that point, I can tell you that when it happened to me, I could not reconnect via wifi even though the chuck kept flying the aircraft.

Steven Lawyer
7129 G-76 Hwy
New Virginia, Iowa 50210
Steven@LawyerFamily.net<mailto:Steven@LawyerFamily.net>
515-229-5352

s0ftice, 2019-10-27:
Steven, yes I know, not just for you, some more reports as well. Just trying to figure out if marph’s issue is the same or something else.
By the way, when you answer via e-mail - due to your signature - your home address and phone number gets immortalized on the public Internet...

StevenLawyer, 2019-10-27:
Thanks for the heads up.  I’ll leave those off! 😊

marph43, 2019-11-02:
@s0ftice : Sorry, I still owe you an answer. I can not tell you for sure if the Chuck was still running, because I retrieved it 10 hrs after it crashed. But as I mentioned, the additional GPS tracker was able to transmit the position during the whole flight, but it's only 2G and another carrier, than the 4G modem, so the coverage might be different. On the other hand it did not reconnect, during it's descent either, although the coverage should have been sufficient. So my gut feeling tells me, that it was  the same reconnection error all along.

Anyways. Did you discover something in the log?

Nicktk23, 2019-11-02:
i did some test`s (about 6 times) on the ground..and i notice that while in 4G mod, if connection on disco is lost for 10 to 20 minutes ...wifi and 4g mod are not accessible anymore, CHUCK is still runing but you cant do nothing to regain control. Il try to extract log file.

s0ftice, 2019-11-03:
That's very useful. Thanks, I will try that. If I can reproduce it on mine, it will be much easier than log files.

@marph43 not yet.

s0ftice, 2019-11-03:
@Nicktk23 I can't reproduce your test. I tried the following
- Disco and SC2 end-to-end connected over 4G - FreeFlight Pro showing the video stream (on the ground).
- I simply turned off SC2
- From my PC, which is connected to the Disco's Wi-Fi, I'm pinging the Disco to see if the Wi-Fi interface is still up (192.168.42.1)

After 1 hour, the Disco is still reachable.
At the same time, I also had a telnet session on the Disco open (PC via Wi-Fi) to interactively see the ulogcat - maybe this session prevented the interfaces from going down?

Nicktk23, 2019-11-03:
after your last post i did the test once again , i give more details about the test run.
1st step : plug the power from battery and insert 4G modem Huawei 3373h into CHUCK useing OTG cable

2nd step : turn on SC2 and connect normaly on Disco`s WiFI.
3rd step : double press Settings Button on SC2...after few moments SC2 and Disco are connected (as expexted)

4th step : double press Settings button on SC2 to switch back on WiFi mod
5th step : turn off SC2....after  14 minutes Huawei e3372h was blinking (showing is disconnect from 4G network) , on WiFi Network scan Disco Parrot wasn`t availabe anymore.

Maybe these steps helps .. work on my parrot every time (sadly).

s0ftice, 2019-11-04:
Why the 4th step? That would not be a real-life scenario, as when you lose 4G when flying, Wi-Fi will hardly be in range to reconnect immediately.

Can you repeat your test with out 4. ?

I will repeat mine without a telnet session.

s0ftice, 2019-11-04:
I have been following your steps and after 30 minutes, I can still see the SSID in a Wi-Fi scan and the modem shows solid LED.
I also ran your step 4, even though it does not make much sense to me as explained above.
I also did not even enable telnet with the double-press on CHUCK's power button in today's test.

Maybe someone else wants to try?

tuttoto, 2019-11-21:
> after your last post i did the test once again , i give more details about the test run.
> 1st step : plug the power from battery and insert 4G modem Huawei 3372h into CHUCK useing OTG cable and turn Disco CHUCK On
> 
> 2nd step : turn on SC2 and connect normaly on Disco`s WiFI.
> 3rd step : double press Settings Button on SC2...after few moments SC2 and Disco are connected (as expected in 4G mod)
> 
> 4th step : double press Settings button on SC2 to switch back on WiFi mod
> 5th step : turn off SC2....after 14 minutes Huawei e3372h was blinking (showing is disconnect from 4G network) , on WiFi Network scan Disco Parrot wasn`t available anymore.
> 
> Maybe these steps help`s .. this glitch works on my parrot every time (sadly).
> 
> PS. i didnt keep alive a Telnet session.

It seems you have everything working, but manage to obtain disconnection from the Disco (after a while, every time) while other users are not... do you have Glympse installed and active? (Or anything else able to keep the connection active?...).
Could you redo your test with Glympse and push notifications tool?

tuttoto, 2019-11-24:
> I have been following your steps and after 30 minutes, I can still see the SSID in a Wi-Fi scan and the modem shows solid LED.
> I also ran your step 4, even though it does not make much sense to me [...]

Ok... I made the try... no telnet enabled. But I have the glimpse and push notifications options installed.
I followed the steps (including the 4th), and after 90mn was still receiving glimpse updates, my disco modem light was still solid, and I reached 12% on the Disco battery. Then, I started the Skycontroller2 and the hotspot on my Samsung S6 edge+, achieving : solid blue light as usual, controller seen in Auriga/FreeFlight, BUT no telemetry and no video streaming...
I didn’t touch the disco but switched the Skycontroller2 to WiFi, which gave me back telemetry/video streams.
Then I switched the Skycontroller2 to 4g and obtained again solid blue light, but this time I had telemetry and video streaming as usual and everything was working well.
My thoughts: the step 4 made the disco looking for WiFi and not 4g, while being visible for zero tier. Until the Skycontroller2 discussed with it in WiFi and told to it to go looking for data on the 4g subsystem. I can confirm that the 4g modem remained connected during the whole test, maybe because it saw no reason to disconnect or maybe because glimpse kept it “alive”.

BigBear90, 2020-05-19:
This happened to me too today... Luckily It was not very far and somehow I managed to get a WiFi connection and land without video feed (controls worked even if video didn't)

BigBear90, 2020-05-19:
First I lost video connection on 4g, but I managed to guide it back home visually, put it in loiter mode and switched to WiFi. That didn't work so I tried to connect back to 4g which gave me back access to the controls. Moved to a better area in case it autolamded and tried to connect back with WiFi. After a few failed attempts and going back and forth between 4g and WiFi I managed to get a decent connection so that I could land it without any problems (still no video feed though) I had to close the freeflight app from memory and when I opened it again it kind of worked for a bit, then lost video feed) after I landed and rebooted the disco everything worked fine.

s0ftice, 2020-05-19:
that sounds quite different from than what the guys described above.

have you tried to tap the back icon (arrow left) in FFpro and the open the flight mode again. this will re-request the video stream from the Disco and fixes it most of the time.

BigBear90, 2020-05-19:
Yes I tried that and that is usually how I get the video feed after connecting to 4g ( it says that the disco is connected but no video feed appears until I either close and open the app or hit back a couple of times). It's been like this from the beginning. Now that you mention it it does seem different from the other descriptions. I'll fly again and see if this happens on a more regular basis. Thanks for the help! :D

s0ftice, 2020-05-19:
Ok, let me know. Probably easier in the Slack channel.
Yes, the guys above were not able to reconnect (also not controls) - neither 4G nor Wi-Fi.

reddai75, 2020-08-15:
Hello, almost the same problem here. Long story short: My parrot took off for a wifi flight (testing a battery mod, 5200mah) from an open field. I have lost connection almost immediately after take off. After that it started loitering for about 10 minutes. I have tried everything to regain connection but nothing worked (switched off and on the remote, switched to 5g then wifi, looked for the DISCO AP with my phone, nothing) after some loitering it probably decided to land but it crashed 400 meters from the take off point (WHY??) at full speed. It "hard landed" on the roof of a house (you can imagine how happy were the owners) I don't think I will be able to get it back (what remains actually) I'll try tomorrow morning ... I'm not sure how to get on the roof, it's 10 meters high with no access.

s0ftice, 2020-08-16:
I think your issue is something else. If you lost connection, it would no t loiter for 10 minutes, it would go into circular landing. I assume you lost video stream, but the connection must have still been up, so technically you could have controlled the Disco using SC2 to bring it down for landing...

reddai75, 2020-08-16:
> I think your issue is something else. If you lost connection, it would no t loiter for 10 minutes, it would go into circular landing. I assume you lost video stream, but the connection must have [...]

It may be it is something else, but it was definetly not connected, the controller blinked red almost immediately after take off. I tried everything: I even switched the remote on and off. I recovered CHUCK and the battery, I've lost the E3372 btw (I left it connected because my plan was to test the new battery in wifi land, and take off in 4g). I downloaded the video and it stops right after take off (that's weird isn't it ?). If you are interested, tomorrow I can have a look to the logs (I've never accessed those files, I have to look how to do it but I imagine it's not rocket science)

federikk, 2020-11-16:
Hello Marc, and thank again for the work

SC2:
root@mppboard:/ # ulogcat | grep uavpal
N boxinit (init) : starting 'uavpal'
I uavpal_sc2 (ulogger) : === Loading uavpal softmod 1.3 ===
I uavpal_sc2 (ulogger) : ... detected Skycontroller 2 (platform mpp), firmware version 1.0.9

DISCO:
/ # ulogcat | grep uavpal
I uavpal_drone(ulogger) : Huawei USB device detected (USB ID: 12d1:1f01)
I uavpal_drone(ulogger) : === Loading uavpal softmod 1.3 ===
I uavpal_drone(ulogger) : ... detected Parrot Disco (platform evinrude), firmware version 1.7.1
I uavpal_drone(ulogger) : ... trying to use kernel modules compiled for firmware 1.7.0
I uavpal_drone(ulogger) : ... loading tunnel kernel module (for zerotier)
I uavpal_drone(ulogger) : ... loading USB modem kernel modules
I uavpal_drone(ulogger) : ... loading iptables kernel modules (required for security)
I uavpal_drone(ulogger) : ... running usb_modeswitch to switch Huawei modem into huawei-new-mode
I uavpal_drone(ulogger) : ... detecting Huawei modem type
I uavpal_drone(ulogger) : ... detected Huawei USB modem in Hi-Link mode
I uavpal_drone(ulogger) : ... unloading Stick Mode kernel modules (not required for Hi-Link firmware)
I uavpal_drone(ulogger) : ... connecting modem to Internet (Hi-Link)
I uavpal_connect_hilink(ulogger) : ... bringing up Hi-Link network interface
I uavpal_connect_hilink(ulogger) : ... requesting IP address from modem's DHCP server
I uavpal_connect_hilink(ulogger) : ... setting eth1's IP address to 192.168.8.100
I uavpal_connect_hilink(ulogger) : ... setting default route for 192.168.8.1
I uavpal_connect_hilink(ulogger) : ... connecting to mobile network using APN "ibox.tim.it" (configured in the Hi-Link Web UI)
I uavpal_drone(ulogger) : ... enabling Hi-Link DMZ mode (1:1 NAT for better zerotier performance)I uavpal_drone(ulogger) : ... setting Hi-Link NAT type full cone (better zerotier performance)
I uavpal_drone(ulogger) : ... querying Huawei device details via Hi-Link API
I uavpal_drone(ulogger) : ... model: E3372, hardware version: CL2E3372HM
I uavpal_drone(ulogger) : ... software version: 22.333.01.00.00, WebUI version: 17.100.13.112.03
I uavpal_drone(ulogger) : ... applying iptables security rules for interface eth1
I uavpal_drone(ulogger) : ... starting connection keep-alive handler in background
I uavpal_drone(ulogger) : ... public Internet connection is up
I uavpal_drone(ulogger) : ... setting DNS servers statically (Google Public DNS)
I uavpal_drone(ulogger) : ... setting date/time using ntp
I uavpal_drone(ulogger) : ... starting Glympse script for GPS tracking
I uavpal_drone(ulogger) : ... starting zerotier daemon
I uavpal_glympse(ulogger) : ... reading Glympse API key from config file
I uavpal_drone(ulogger) : *** idle on LTE ***
I uavpal_glympse(ulogger) : ... disabling Glympse, API key set to ignore
E boxinit (init) : untracked pid 1261 'uavpal_glympse.' exited

s0ftice, 2020-11-16:
Federico, this is just the the standard log from now, that does not help.

You would need to enable debug mode (second part of https://uavpal.com/disco/faq#debug). Only then, your log files will be persisted. So if it happens again, you need to pull that (/data/ftp/internal_000/Debug/ulog_debug_{date/time}.log) and send it to me/post it here. There is no way to get the log from last time, if the debug mode was not enabled or the drone has been restarted/shut down.

federikk, 2020-11-18:
Thanks Bjorn,
i've activated the permanent log but not managed to have the loss of communication again (luckiyl). Can i leave it on indefinitely?

the only thing that went different that time, the disco asked for a calibration on start up, which went flawlessly. i don't know if can be a hint for you

thank you
F

s0ftice, 2020-11-19:
Just leave it on.
Calibration requests seem random to me.

softice (not Bjorn) 😉

parrotdisc, 2021-02-04:
Hi Guys it happened to me also today.
I was at home and had just finished the installation of the software.
I have tried the 4g connection 5 times , 4 times everyting worked well but one time this happened :
After gaining the 4g connection with the videofeed at a certain point for no reason i have lost the videofeed, what I have noticed is that the green light of the huawei modem was blinking like if it had lost connection. For sure 100% the 4g coverage was there  because the parrot was on the floor and didn't ove when it happened.
Have tried to connect through normal wifi by ckicking on the remote settings botton but nothing happened. Tried also switching the remote off and on again but it would just blink red. . There was no way to get the connection back , the parrot was like frozen with the modemlight blinking. I have noticed that the model was quite hot. I use an Huawei E3372-153.
Did anybody have this issue using a different type of modem?
After switching the parrot off and on again it was ok. 
Hope this helps. 
It would be great to find out whats causes this issue.
Thanks!
R.

AndKe, 2021-08-19:
@s0ftice  I will make a debug log for that within a few days.

What I observed is this:
Powered up, switched to 4G - which worked as expected.
To simulate some total 4G failure, I pulled the USB cable from the E3372
Connection got lost, SkyController2 flashed red for minutes, no fallback to Wifi.
I tried to force the controller to Wifi by double-pressing the setting button. - no success.
I reconnected the E3372 (not receiving a Glympse SMS tells me this was not possible and that the device was not re-initialized)
I switched off the SkyContoller2 , then on - again - it could not re-establish wifi connection.
Disco could be shutdown using power button.
This tells me that in case a USB cable, or 4G device failure, the plane cannot be recovered by falling back to WiFi.

AndKe, 2021-08-19:
@s0ftice  Preparing for the promised log-gathering I discovered that I already did enable logging before.
I packed some other directories that may have something useful as well.
The log attached, if the very day/situation I described.
[DiscoLog.zip](https://github.com/uavpal/disco4g/files/7017481/DiscoLog.zip)
I hope this helps you understand the problem, feel free to request more logs if needed.

mainframe, 2021-08-20:
@s0ftice I remember that this issue was real with older disco4g versions - ie if 4g modem usb disconnect happened switching to wifi did not allow to connect. I don't recall all the details - but it might have been related to zerotier routing (could have been that traffic was still directed to ZT tunnel - so no valid backroute for wifi connection - ie no proper reset on 4g modem usb disconnect).

AndKe, 2021-08-21:
The 4G mod version is disco4g-1.3
E3372 softwarte version 22.333.01.00.00     webui 17.100.06.00.03

I tested an area with marginal 4g - where I could easily do a proper "loss of signal" 

Log1: 
started up, switched to 4G , 
pulled modem and re-inserted two times, no automatic fallback to wifi, and no way of regaining 4g connection.

Log2:
This was just a reboot og Disco only, where I was checking whatever it would just reconnect to the controller that was in 4G mode all along.

Log3:
started up, switched to 4G , then carried the Disco out of cellular range .
it did not fall back to wifi.
I carried the controller so it too lost the internet connection, 
the Disco did still not fall back to wifi.
Pressed settings button twice, and wifi worked fine.

Isn't the fallback to Wifi supposed to work?
[ulog_debug_2021082115.zip](https://github.com/uavpal/disco4g/files/7025816/ulog_debug_2021082115.zip)

AndKe, 2021-08-28:
Here comes another set of debug logs, this time, most logs display exemplary 4G error handling, things going as expected  - that is until the last log.
The last log, I am again unable to establish wifi connection - then the Disco apparently crashes, just shuts down(power off)
[ulog_debug_202108231.zip](https://github.com/uavpal/disco4g/files/7071173/ulog_debug_202108231.zip)

-Scary.

Robifpv20, 2021-10-04:
Hello Disco pilots.

I have been flying a bit on 4G LTE lately and always ended well and without problems. Although the video preview disappeared many times in one day, despite the fact that I had the control and had to restart the application several times until the image from the camera appeared. But I was able to steer the drone all the time and landed smoothly.

I always fly on an indirect connection (glympse displays the letter R)

Now be careful what I discover. Despite the fact that we have a connection via LTE, at least for me Disco still sends a Wifi signal that can be connected to.

And now very important, but I checked it only during wi-fi flights:

If we take another smartphone or tablet and connect to Disco via wi-fi and run the Freeflight Pro application, when the skycontroller temporarily loses connection, the second phone will take over the connection with the drone and will not allow to connect again with the Skycontroller. We will even have a camera preview.

The call will be blocked all the time until the other phone hangs up.

Only then will skycontroller connect to Disco and regain control.

Did you know about it?

Disco then has no connection with the pilot and activates RTH and automatic circular landing. The LED on the controller will flash red all the time.

The only hope is that Disco will get away from the other phone and lose its wi-fi connection, then the skycontroller will automatically connect in two seconds and we will be able to control.

Robifpv20, 2021-10-04:
Just connecting your phone, tablet or computer to Wi-Fi Disco does nothing. Only turning on the Freeflight Pro / Auriga Bebop application initiates the connection and allows the smartphone to take control and block the skycontroller. I also had some successful long distance flights on LTE as I was returning after 1 hour, even without a single disconnect, only slight video clipping sometimes. Orange Polska network in the Huawei modem and on the phone. But when someone called me during the flight, the LTE network on my smartphone changed to H +, which caused the hotspot to restart and disconnect from Disco. That's why now I use a tablet to fly and no one calls.

Robifpv20, 2021-10-10:
> Oto kolejny zestaw dzienników debugowania, tym razem większość dzienników wyświetla przykładową obsługę błędów 4G, wszystko idzie zgodnie z oczekiwaniami - do ostatniego dziennika. Ostatni dziennik, znowu nie mogę nawiązać połączenia wifi - wtedy Disco najwyraźniej się zawiesza, po prostu wyłącza się (wyłącza zasilanie) [ulog_debug_202108231.zip](https://github.com/uavpal/disco4g/files/7071173/ulog_debug_202108231.zip)
> 
> -Straszny.

Hello friend, see what I wrote in the last two messages regarding this issue. This is what it looks like in my Disco model. Sometimes shorter, sometimes longer, but almost every time CHUCK finds a signal from the Skycontroller. Even then, if I temporarily turned off the wifi hotspot on the tablet that I use for video preview.

s0ftice, 2022-02-12:
this seems to be related to the microSD card in the modem. Please try without and let me know if that solved the issue.
Thanks!

AndKe, 2022-06-25:
@s0ftice : I've never used the SD slot of the modem.

Robifpv20, 2022-06-27:
It seems to me that the memory card in the modem has no effect on blocking the connection and on disconnections. I often recorded to the card in the modem and did not notice any problems. Recently, I was disconnected when I was flying LTE, because there was a storm in the town next to the cellular network transmitter. Disco came back by itself and started landing very nicely against the wind. But I was able to switch to wi-fi at the last minute and take control, and then switched back to 4G LTE while in the air.

disco555, 2022-08-04:
Hi All , Can anyone help me , i set up the 4g softmod parrot disco got a solid blue light when into Zerotier the disco and controller are bridge, the next day decided to do a test flight with the 4g softmod in the park , turned on the disco then the skycontroller 2 got the green light  everything was connected then  i hit the setting button twice and the light went purple/pinky/ flashing blue for ages then went back to red and then to green 
i went home and tried this again and it worked  at home i got a solid blue light and the Zerotier  was bridge 
I am using a Huawei p20 phone is there something i am missing ?

s0ftice, 2022-08-04:
> Hi All , Can anyone help me , i set up the 4g softmod parrot disco got a solid blue light when into Zerotier the disco and controller are bridge, the next day decided to do a test flight with the 4g [...]

this is not related to the above described issue. You got some useful answers in the Disco Group on Facebook, I suggest you go trough that first. Have a look at https://github.com/uavpal/disco4g/wiki/faq#badconnection as well.

Robifpv20, 2022-08-07:
If the LTE connection is unstable or too slow, it will disconnect frequently. First, check the LTE signal strength and internet speed in your area, if the 4G LTE Internet is good, you should not experience disconnection. The blue diode should be on all the time, and if during the flight a disconnection occurs and the red diode is blinking, it should take several seconds to quickly switch to blue. As a colleague said above, go through all the threads on this topic and gain the required knowledge and experience later. I greet you, do not give up.

Leolux226, 2022-10-30:
Hello im new here and have exactly same issue i tested a lot around looks like really happend only in 4g mode mostöy after 20-40 minutes wifi mode it runs 1h or more without problems. i checked if overhets but with open cover same if this heppend the check stops record files on sd crad are like not finished container wifi connection back not possibel only after restart the disco. I use 1.41 master version after glympse was not working, but btw glympse still now not work right dont show the right way. strange is teh plane still work my flight data on the parrot page is compleed it looks like a part of the check fails out. It happend on bench to connection lte is really good. what im checking now is OTG cable and probebly the stick at self. Is there any news about this problem i love this plane i love the 4g mod i buyd batterys 2 planes and got stuck in this issue for now

Leolux226, 2022-10-31:
So i did it agan i truy cool down the lte stick took longest test possibel 50 min but still get the issue after i made debug on take sd card out of the stick and i got it agan i will attach the log here i gues the problem comes at line 31129 at the log hopefully this will help to fix
[ulog_debug_20221030232151.log](https://github.com/uavpal/disco4g/files/9897106/ulog_debug_20221030232151.log)

Leolux226, 2022-10-31:
i guess thats the event looks like usb kernal issues i will truy other otg cable and i will order a other lte stick that son the supported list
LOG:
U I MassStorage (dragon-prog/Behaviour)          : List id = 0 ; name = "internal" ; internal = 1 ; size = 29009 MB ; occupied = 5485 MB; 
U I COMMANDS    (dragon-prog/Behaviour)          : Cmd Tx : mass storage state ID <0> name <internal>
U I COMMANDS    (dragon-prog/Behaviour)          : Cmd Tx : mass storage state info ID <0> usage <5485/29009> plugged <1> full <0>
K E KERNEL                                       : ci_hdrc ci_hdrc.1: fatal error
K E KERNEL                                       : ci_hdrc ci_hdrc.1: HC died; cleaning up
K E KERNEL                                       : usb_wwan_indat_callback: resubmit read urb failed. (-19)
K I KERNEL                                       : usb 1-1: USB disconnect, device number 2
K I KERNEL                                       : usb 1-1.1: USB disconnect, device number 3
K E KERNEL                                       : option: option_instat_callback: error -108
K W KERNEL                                       : bcm_rpc_call_with_return: bcm_rpc_tp_recv_rtn() failed
K W KERNEL                                       : bcm_rpc_down
K W KERNEL                                       : DONGLE rpc dump:
K W KERNEL                                       : Version: 0x6257240 State: 4
K W KERNEL                                       : session 105 trans 0x3ab9 oe_trans 0x5f75 rtn_trans 0x0
K W KERNEL                                       : xactionID out of order 0 reorder ovfl 0 dropped hole 0
K W KERNEL                                       : reorder queue depth 0 first ID 0x0 reorder_q_depth_max 0, tossthreshold 40
K W KERNEL                                       : rpc_dbus_state_change: DBUS is down
U E VIDEO_CAM_ANGLES(dragon-prog/camAngles)      : [video_camAngles->handler_thread_video_camAngles()] error, failed to get intput data (res=-1)
K E KERNEL                                       : ci_hdrc ci_hdrc.1: force halt; handshake ee400144 00004000 00004000 -> -110
K E KERNEL                                       : ci_hdrc ci_hdrc.1: HC died; cleaning up
U E VIDEO_REPROJ(dragon-prog/video_reproj)       : error, video_imu_frame_sync2 returned -2
K W KERNEL                                       : bcm_rpc_down
U E VIDEO_REPROJ(dragon-prog/video_reproj)       : time between two frames to reproj > 50ms : 300296µs

ClassAxion, 2022-11-02:
@parrotdisc so do you mean, that only modem without hilink (web interface) firmware cause the problem? You can flash hilink firmware on that modem and check if problem no longer exists.

[29 more replies]