uavpal/disco4g #65: Anafi LTE mod

nlona, 2019-09-28:
As Anafi does nota has chuck connection, this mod should be use USB rear port on the anafi and sc3 could connect

SuperJMN, 2019-09-28:
I can't wait for this mod to become a reality

s0ftice, 2019-09-30:
What is a CHUCK connection? :) CHUCK is the flight controller of the Disco and has nothing to do with the Anafi.

An Anafi mod is not as simple, as unlike the Disco and Bebop 2, it’s a closed platform. There is no system access allowed and Parrot put in quite a lot of controls that it stays that way (following DJI’s example). Unless someone finds a software exploit, there is no “softmod” for the Anafi in sight.

However, an Anafi 4G mod would also be theoretically possible with some extra hardware on the drone side (e.g. Raspberry Pi Zero) and without having to exploit the software. It won’t be as “user-friendly” as the Disco/Bebop2 softmod though. If there’s any progress, I will update here.

nlona, 2019-10-04:
Sorry I meant connection top Chuck. Know that anafi has not chuck, telnet connection should be easy to test if present,

s0ftice, 2019-10-04:
Powering the modem from the Anafi is not the issue.
As mentioned before, it’s a closed platform and there is no system access. In other words: no telnet.

nlona, 2019-10-04:
As Anafi should use Ambarella chipset could give a clue on how to solve https://www.goprawn.com/forum/ambarella-cams/1536-getinfo-script-for-a12-cams

closed 2019-10-04 by nlona