uavpal/disco4g #67: Parrot Disco Constant Rebooting After Downgrade From 1.7.1 to 1.0.5

Drobny82, 2019-10-18:
Hi everyone. I have BIG PROBLEM. Can anybody help ? So i am a Disco owner few months now. I try flight in default (get from box and start fly) and did it is awesome toy ! But i decide to get UAVPAL in it. My Parrot is (was) 1.7.1 ver and Skycontroller 2 is 1.0.9 (silver joysticks with white line on antenna). So in the beginning i found a movie on YT: 

**https://www.youtube.com/watch?v=vKB7fNwb4ps** 

I try to do exactly the same as on movie. But Disco was not connected to my ZeroTier network. Only Skycontroller worked very well and was connected. Then i go to the link below the movie description and i found GitHub site :) So after hours of try do exactly as on:

**https://github.com/uavpal/disco4g/wiki/Installation.** 

Situation was still the same. Skycontroller worked well and was connected but Disco not. So i decide to downgrade both. First i try did Skycontroller to downgrade i did same thing as here:

**https://github.com/uavpal/disco4g/wiki/Manual-firmware-upgrade-and-downgrade-of-Parrot-Skycontroller-2** 

But i was confusing on the end exactly there:

 **/data/ftp/internal_000/adb shell
/usr/bin/pinst_trigger 1
reboot && exit
rm /data/ftp/internal_000/adb /data/ftp/internal_000/mpp_update.plf**

So after **reboot && exit** i went back to CMD and i turn on telenet on disco (2 x disco power button) and i copy and paste this line:

**telnet 192.168.42.1**

And then last line:

**rm /data/ftp/internal_000/adb /data/ftp/internal_000/mpp_update.plf**

And i did **reboot && exit** again. So my controller was still same firmware 1.0.9 and i still could connect it with my ZeroTier network. Then i try to downgrade my Disco manually from 1.7.1 to 1.0.5 because when i try push power button 4 times (4 hold) as on this movie from YT:

**https://www.youtube.com/watch?v=Fpqxa6wLkq0**

Then like on this movie 3 times and hold:

**https://www.youtube.com/watch?v=btgUetj9Lnw**

Just nothing happend. On the comments below i found info about ver. 1.7.1 that does not work with this ver. of Disco and i need to do this manually. So i go to this site:

**https://github.com/uavpal/disco4g/wiki/Manual-firmware-upgrade-and-downgrade-of-Parrot-Disco**

So this step was really easy because after hours of trying everything before i was closer to understand everything well :) But in the end when Disco should reboot 3 times **(Wait for about 3 minutes while the Disco firmware gets installed (the Disco will reboot 3 times)** my Disco start to reboot constantly and blue light is constant to and rebooting never stop :) Even after 20 min and more. So i cant even go to connect it to my WI-FI to go there with FileZilla and even when i try to connect it to PC with USB micro cable nothing happend. After everything i even try this with ver. 1.7.0:

VIA A MICRO USB FLASH DRIVE 
1. Connect your micro USB flash drive (not provided) to a computer. Make sure that your flash drive does not contain any files in .plf format. 
2. Go to the Parrot Disco’s support page on www.parrot.com and download the update file available in .plf format. 
3. Copy this file to the root of your micro USB flash drive. 
4. Turn on the Parrot Disco. 
5. Connect the micro USB flash drive to one of the Parrot Disco’s ports.  > The Parrot Disco automatically restarts and the warning light flashes yellow. Then, the Parrot Disco restarts again. 
6. Wait until the Parrot Disco has completely restarted. 
7. Disconnect the micro USB flash drive. > The update is complete. Make sure the new version has actually been installed: in the FreeFlight Pro application, go to Menu > Settings. The version number is shown in the general information.

My Disco reboot every 22 sec and do this every time when i turn it on. I do not know what i need to do now because now i cant fly my Disco even in default without UAVPAL. If anybody can help i will be very grateful.

Drobny82, 2019-10-19:
When i push power button 3 times and then 4 hold my LED light turn on to blue then LED turn off for a second and then start to be steady green. And from this moment nothing happened. When i turn Disco off by pushing one time power button when LED i steady green Disco came back to constant reboot. When i push power button few times (4) between 22 sec reboot Disco just turn off. And to turn on it again i need to plug off battery and plug in again.

s0ftice, 2019-10-21:
Hi Drobny82
The reason for the indent between `reboot && exit` and `rm /data/ftp/internal_000/adb /data/ftp/internal_000/mpp_update.plf` actually detonates that the first command is ran on SC2 and the second on the Disco. By just simply following the instructions 1:1, nothing can go wrong as the reboot kicks you out of SC2 automatically, so you "end up" on the Disco again where you use the "rm" command to remove the temporary file.

Why did you downgrade your Disco from 1.7.1 to 1.0.5? https://uavpal.com/disco/sub#requirements states clearly "Parrot Disco / Parrot Disco-Pro AG with firmware **1.4.1 to 1.7.1**", 1.0.5 will not work with the softmod.

Why did you try for hours without joining the community on Slack and simply asking for help? https://uavpal.com/slack we have 1000+ members... The link is placed on the main Github page as well on the Installation Wiki page.

I don't think I can help you with this particular issue, never tried such a low firmware version. However, the download link provided for 1.0.5 is from the domain firmware.parrot.com and thus 100% genuine.

You can try to reset the Disco: To reset Parrot Disco, make two short presses and then a long press (about 10 seconds) on the on/off button of your Parrot Disco.
The light of the on/off button flashes orange and red and Parrot Disco switches off.
To complete the reset process, switch on Parrot Disco.

Good luck!
SoftIce

Drobny82, 2019-10-21:
Hi. Thanks for answer. First reason to downgrade of Disco and SK2 was to upgrade it again to try install softmod again because in the first time i just did something wrong. Second i thought i can do this without any help when i just follow instruction 1:1 but i was wrong because i did not understand few thinks but i decide to just try it because i thought i can easily downgrade Disco and SK2 so i was wrong again. Reset just not work in my case i think i need to buy new CHUCK and try it again but this time with help to do everything well. Thanks for link to the group ill join there. When ill find good pre owned CHUCK ill ask later about few thinks. Thank You again.

closed 2019-10-24 by s0ftice

Peterthegreat, 2022-09-13:
@Drobny82 do you still have the CHUCK? I can repair it

Drobny82, 2022-09-13:
> @Drobny82 do you still have the CHUCK? I can repair it 

Sold it few months ago. But thanks for answer.

AlgirdasPro, 2023-02-08:
I have same problem how can you repair it?

Peterthegreat, 2023-02-08:
@AlgirdasPro Sure. I specialized in CHUCK repairs

AlgirdasPro, 2023-02-09:
> @AlgirdasPro Sure. I specialize in CHUCK repairs.
> 
> EDIT: Oh, I just took a better look at your question. I am sorry but I spent quite a lot of time figuring how to do it. I only sell the repair service. I can repair it remotely so you won't spend extra money on shipping

How can i contact you?

Peterthegreat, 2023-02-09:
[image]