uavpal/disco4g #68: Led blue flashing (no bridge established)

seboss1111, 2019-11-17:
Hey guys ! ( i'm a engineering student passioned about drones <3 )
I've an issue for a while now... 
I'm not able to established a bridge connection with ZeroTier : my sc2 is blinking blue..

BUT the weird thing (probably related to my issue) is that it is WORKING SOMETIMES 👍 
(SOLID BLUE : But i'm not able to get a video feedback or any feedbacks from the drone with freeflightpro, and then, after 2minutes, it disconnects and goes back to wifi mod)

Of course : 
-in ZeroTier, both devices are "ONLINE" with bridge enabled
-everything was configured (ssid, pass, APN, etc...) : not surprising if i can get solid blue sometimes
-in wifi it's working properly (feed backs)

I really need help here, i won't be able to solve this huge issue alone... 
I can get you every infos you need.. just tell me what you need and what to do !  ;)
Thank you all in advance 💯

s0ftice, 2019-11-18:
Hi seboss, zerotier reduced their network of relay nodes drastically in August. And as the FreeFlight Pro/SC2 connection is very sensitive to latency, you most probably get disconnected before the video stream even starts. What carrier are you using? I assume you are getting a relayed connection. If you can manage to get a direct connection, this issue will not occur.
See also: https://uavpal.com/disco/faq#ztdirect and https://uavpal.com/disco/wiki/Known-Working-Mobile-Carriers-and-Settings

SoftIce

seboss1111, 2019-11-19:
I've pretty much tested everything now...
And unfortunately my carrier is here on your list : "Bouygues telecom" in France..
Yeah I'm quite sure that it is ZeroTier the problem here.. :/

Any idea on using something different? 💛 
Even with an old raspberry pi zero, as long as it would work for me, I'm ready... Tell me... 👍

tuttoto, 2019-11-20:
Hum... it works perfectly for me with Bouygues in France. BUT, it seems the apn I use is not the one listed in github... I used the apn given by Bouygues (something like mmsbouygtel.com) and made sure to have good 4G level on the phone. I have 2 B&you sims with this same apn, then glympse reports are telling “direct connection “ and 50ms latency... I posted my maiden flight with 4G on YouTube, then I can swear it works.

seboss1111, 2019-11-21:
Yeah I discovered that too while searching infos on my network provider, under network configs in my phone (apn : mmsbouygtel.com)
I have not "B&You" but "Sensation" offer on my phone (with the free multi-sim option the second for the sim in my drone : the two basically share the same internet offer --> maybe that's the problem???)
Anyway you give me hope thanks a lot @tuttoto !!! 👍

s0ftice, 2019-11-21:
it could be. if you can borrow another SIM (different plan) for a quick test that would help to find out.

tuttoto, 2019-11-21:
I investigated sensation and b&you contracts (only marketing differences, the technical infrastructure and parameters are identical), and found out that apn for phones are “mmsbouygtel.com” BUT apn for the second sim (when subscribing the “multi-sim” option) is “ebouygtel.com”.
I have 2 “phones” SIM cards (allowing voice/sms/mms/data), one in my phone, one in the disco... with “mmsbouygtel.com” for both.
You have 1 SIM card identical to mines , but your second one is not allowing voice/sms/mms... only data, and “ebouygtel.com” as apn...

youness103, 2019-11-23:
Bonjour ça fais plaisir de voir des français ici bon j'ai rencontre le même problème j'ai 2 carte SIM Bouygues une première normal dans mon téléphone et la deuxième pour les données dans les données j'ai changé de forfait rien que pour ça est ça me marche pas. 
Avant j'ai essayé avec 2 carte SIM free ça marche. 
Du coup j'ai êtes acheter une carte SIM free pour mon disco, ça marche pas. 
J'ai supprimer l'APN de Bouygues dans mon téléphone j'ai créé un avec le même APN que le disco je nomes les 2 ( libre) puis ça marche. 
La ou je suis la connexion êtes très très merdique j'attend des meilleurs conditions pour un test vol avec une meilleure couverture 4G.
Merci à vous

youness103, 2019-11-23:
J'ai l'impression que plusieurs mots sont transformer après publication

youness103, 2019-11-23:
Sûrement à cause de la tradition

youness103, 2019-11-23:
Libre = opérateur FREE

tuttoto, 2019-11-24:
Félicitations pour avoir résolu ton problème, je regarde les cartes de couverture 4g avant de choisir mes trajets... j’espère que seboss1111 pourra résoudre son problème aussi.
Congratulations for having solved your issue, I check 4g maps before choosing my fly paths... I hope that seboss1111 would solve its own issue too.

youness103, 2019-11-26:
help me
my problem is that once the connection shares from my pocophone the SC2 does not connect:
"no apparatus connected" on ZEROTIER the disco is online:
the SC2 is difficultly connectable issue; is there a way to connect the SC2 via USB cable to the network ???????

youness103, 2019-11-26:
I use a smart telecom chip a normal chip in my pocophone the other in the modem dedicated to the connection in the latter the APN "ebouygtel.com".
I created a camera in my pocophone nome "ebouygtel.com"
it does not work

s0ftice, 2019-11-30:
Yes, you can use a USB to Ethernet cable on SC2. But this is only required to analyse/investigate issues.

Can you try to set the APN to "mmsbouygtel.com" on both sides?

closed 2020-03-23 by s0ftice

skyer29, 2020-05-08:
> Hey guys ! ( i'm a engineering student passioned about drones <3 )
> I've an issue for a while now...
> I'm not able to established a bridge connection with ZeroTier : my sc2 is blinking blue..
>
> BUT the [...]

I have the same problem.
Blue led flashing  and on zerotier both device are on line.  The problem is on the softmod on the disc because I tried ski2 with another disco and works very well.  Help me please😊

s0ftice, 2020-05-09:
Please join us on Slack, easier to discuss https://uavpal.com/slack