uavpal/disco4g #69: Sky Controller never shows up on zero tier

jimschuler3, 2019-11-20:


jimschuler3, 2019-11-20:
I double press settings button 
i first upgraded to 1.7.1 disco connects to ZT but don't see SC
so i downgrade firmware to 1.4.1 same issue 
i double press disco to activate ZT still nothing
using Huawei e3372h 510
tmobile sim 
apn 
epc.tmobile.com
im in the U.S.
Does it have to be the same apn on my tmoble iphoneX
i looked on the usb modem and it said epc.tmobile.com so thats what i put in the bin file
any help would be appreciated

jimschuler3, 2019-11-20:
i just removed my ZT network and started a new one  ....reinstalled softmod with newZT# and its the same thing

s0ftice, 2019-11-21:
what color is the LED of your Skycontroller 2? Flashing masgenta and then going back to green? What hotspot device are you using?

jimschuler3, 2019-11-21:
iphone x and yes flashing megenta then yellow then red then green

s0ftice, 2019-11-21:
have a look here: https://uavpal.com/disco/faq#wififail

tuttoto, 2019-11-21:
> iphone x and yes flashing megenta then yellow then red then green

I managed to connect with my iPhone 8,  but I had to rename it (I used my surname instead of “surname’s iPhone”...). I have only checked that I had video and steady blue light and headtracking, but I have not tried it in flight.

jimschuler3, 2019-11-21:
just tried that in itunes but  it just says jims iphonex    not "jims iphonex"
im stumped

jimschuler3, 2019-11-21:
ill change the name to just a single word maybe?

jimschuler3, 2019-11-21:
heres a video of what its doing

https://www.youtube.com/watch?v=22Xh8GarCJU&feature=youtu.be

s0ftice, 2019-11-22:
You can try, but you also should ensure that the config file on SC2 gets updated, e.g. via reinstalling the mod.
Can you attach a screenshot of your iPhone's name? Settings -> General -> About -> Name, as well as the current conf/ssid file?

youness103, 2019-11-22:
Bonjour je suis dans le même cas de figure, j'ai beau essayer de recommencer la procédure le disco ce connecté bien mais la SC n'apparaît pas ça clignote bleu 🔵 après ça passe au vert 🌳je suis avec les derniers Version de mise à jour

jimschuler3, 2019-11-23:
Photos of my settings
[image]
[image]

tuttoto, 2019-11-23:
> Photos of my settings
> [image]
> [image]

It would be interesting to have the copy of the configuration file you loaded in the Disco for the Skycontroller...

jimschuler3, 2019-11-23:
[image]
Says  it loaded

tuttoto, 2019-11-23:
I tried with my iPhone 8xl/256gb ios13.1.3... and this time it stays magenta for ever... then I tried to connect with an other phone on the hotspot, and discovered it wasn’t working for it also... I even entered the ssid/password manually (ssid was not even displayed on the list of available networks...) but it didn’t work either...
At some point, I disabled the authorization for non-iCloud users to connect to the hotspot, then enabled it again, which triggered an odd message requesting my approval to enable WiFi for users to connect to the hotspot... I pressed “ok” then my other phone was able to see the ssid and to connect to the hotspot, then after having switched the Skycontroller2 to WiFi and back to 4g mode... after ages... magenta turned blinking blue, then solid blue... video stream was s..t (I think I’m on reduced data rate on this data plan), but strangely glimpse reported direct connection and <>50ms latencies (not same apn as the Disco, though...). I think apple fucked the hotspot somehow... as I have not at all such weird behavior with my old Samsung s6 edge+... when I start the hotspot on the Samsung, the magenta light start blinking blue after seconds, then solid blue , then video stream a few seconds later...
I was not too confident about my iPhone (as I have to restart the WiFi many times during a single evening while watching YouTube (it looks like I’m connected to my home WiFi... but streaming is stopped, until I switch WiFi off/on)), then I never used it to actually fly my disco... I think I will not use it for 4g either. (Or maybe like Dustin Dunnil... he uses an Android phone for the hotspot and an iPad for FreeFlight/Auriga).
I hope it helped... if other users could share their point of view about iPhone/hotspots I would be interested.

s0ftice, 2019-11-24:
> [image]
> Says it loaded

yeah, but I can't see what the conf/ssid file is currently set to on your screenshots

s0ftice, 2019-11-24:
> I hope it helped... if other users could share their point of view about iPhone/hotspots I would be interested.

Least issues with iOS devices as long as users follow the two tips here https://github.com/uavpal/disco4g/wiki/faq#wififail

jimschuler3, 2019-11-24:
Ok i may try to scrap the whole idea of using my iphone 
i see people using there home wifi  is that possible and what do i use as the apn is that going to also be the name of my network

jimschuler3, 2019-11-24:
> I tried with my iPhone 8xl/256gb ios13.1.3... and this time it stays magenta for ever... then I tried to connect with an other phone on the hotspot, and discovered it wasn’t working for it also... I [...]

were do i turn that(authorization for non-iCloud) on and off

tuttoto, 2019-11-25:
> were do i turn that(authorization for non-iCloud) on and off

It is the toggle just above the password field (in the hotspot parameters). To be precise, I switch off WiFi on the control center (when you slide up from the bottom of the screen, where there are sliders for volume, the light, etc) and then also in the parameters (as it is still ON after having put it OFF on the control center... ), THEN I put ON the non-iCloud users authorization toggle in the hotspot menu, which triggers a pop-up box requiring activation of the WiFi, I tap on it... it seems to reset properly the hotspot enough for the Skycontroller2 to connect on it...

s0ftice, 2019-11-25:
> were do i turn that(authorization for non-iCloud) on and off

All you need to do is "Allow Others to Join" under Settings -> Personal Hotspot

s0ftice, 2019-11-25:
> Ok i may try to scrap the whole idea of using my iphone
> i see people using there home wifi is that possible and what do i use as the apn is that going to also be the name of my network

iPhone is no issue, trust me. You can use the Home Wifi, but I strongly suggest to get it to work on your phone first. Ping me on Slack when you're online. I will run you through a set of commands to find out what is currently configured on your SC2.

APN does not change if you want to use Home Wi-Fi, as the APN is set on the Disco.

youness103, 2019-11-26:
aide moi 
mon probleme est que une fois la connexion partage depuis mon pocophone le SC2 ne ce connect pas:
"aucune appareille connecté" sur ZEROTIER le disco est en ligne:
le SC2 est dificilement connectable question ; existe il un moyen de connecte la SC2 par cable USB au reseau ???????

closed 2020-03-23 by s0ftice