uavpal/disco4g #70: 4g mod on sprint modem

okstuv, 2020-02-19:
Hi, great repo. I'm excited to try this out. As a Sprint MVNO existing customer, I'm curious if anyone has tried this using a Sprint-compatible 4g USB stick modem. 

Have you heard of anyone doing this? Do you foresee any issues applying your instructions to a different 4g USB stick modem like this: https://www.amazon.com/Sprint-Wireless-Modem-Broadband-Aircard/dp/B006A9ZWLW/

s0ftice, 2020-02-20:
Hi there.
No it will not work by just following the instructions.
Non-Huawei modems require a different kernel module (“driver”).
This will also require some code changes, which we don’t have on our roadmap. But with some Linux knowledge, anyone can make the modification to support other modems (and hopefully contribute back via PR).

But is it really worth it? As...
1. T-Mobile is now taking over Sprint and will therefore combine the networks in future.
2. In addition, 2G/3G networks (like Sprint’s proprietary CDMA) will go away in future with 5G being rolled out these days.

okstuv, 2020-02-20:
Thanks for the reply. Great points! The kernel point totally makes sense now that I looked deeper into the installation steps. CDMA being on the way out is a pretty compelling reason to not invest too much time in a Sprint-native solution. Cheers!

closed 2020-02-20 by okstuv