uavpal/disco4g #71: glympse developer account creation broken

pkusmierski, 2020-03-04:
Hey UAVPALs!

For over a month now I'm trying to set up the developer account with Glympse, but it always fails with "invalid CAPTCHA". Somebody posted on their FB page about it, there is also stackoverflow thread [1], but they don't seem to care. Is there an alternate site/service that would work?

[1] https://stackoverflow.com/questions/60132935/unable-to-create-a-glympse-account/

s0ftice, 2020-03-04:
Hi there. Thanks, yes we are aware - this topic is being actively discussed on our Slack channel as well. We might look out for alternatives soon. You can also leave out Glympse entirely for now as it's not mandatory.

pkusmierski, 2020-03-10:
I managed to get in touch with somebody through their messanger and account creation works now! I just created my account.

closed 2020-03-10 by pkusmierski