uavpal/disco4g #72: Video stream lost after 25 min. flight

ClassAxion, 2020-03-21:
Hi!
I lost my video stream after 25 min. flight as you can see in attached photo. Has anyone ever had such a problem? This is one of my first flights on LTE so I didn't try to restart the application or sky controller 2. When I switch to WiFi the video stream started working again so it should work if I go back to LTE. 

I'm using latest versions on disco and skycontroller 2 + e3372h with HiLink.

[image]

s0ftice, 2020-03-23:
As you still had a working connection, you can try to press the "<" button on the upper left to go to the main screen and then go back into the video, this should relaunch the video stream.

ClassAxion, 2020-03-23:
I'm trying, and it doesn't work :/

Here you can see: https://youtu.be/D_kLnesOJM8

if I have this situation again I will try in a different way.

s0ftice, 2020-03-24:
Can you reproduce it on the ground? Maybe you did not have enough bandwidth available? If you are using Glympse, did you have a 3G or 4G connection, what was the signal strength shown when the video disconnected?

Also check this: https://uavpal.com/disco/faq#badconnection

ClassAxion, 2020-03-24:
No I can't.. I have many BTS in my region so at 80-100m alt. I have >80% LTE everywhere.
It only disconnects when reconnecting to other BTS.  

Today I have 30 minutes flight without any problems. I turned debugging so if the situation happens again, I will have some logs for you.

s0ftice, 2020-03-25:
Ok sounds good. I have seen some of your videos on Youtube, awesome!!
Why don't you join the #Disco4G community on Slack? https//uavpal.com/slack

SoftIce

ClassAxion, 2020-03-25:
I've been thinking about this and I'm going to join you ;) I really love disco among all mine flying wings.

s0ftice, 2020-05-17:
can I close this issue?

ClassAxion, 2020-05-18:
Yes, we have solution for that and it's not a critical issue :)

closed 2020-05-18 by ClassAxion