uavpal/disco4g #73: 4g no funciona

Pepo234, 2020-04-12:
Hola, como estáis? Saludos a todos los que ayudáis aquí antes que nada. Mi problema es que mi módem Huawei e3372h-153 no conecta con el disco. No se que hacer, no encuentro más opciones. El skycontroller enlaza perfectamente con zerotier pero el disco no enlaza con mi módem. Que opciones tengo? Estoy buscando aquí información de todo tipo pero no he conseguido solucionar este problema. He intentado muchas muchas veces pero no he conseguido nada. Alguien puede ayudar? Muchas gracias.

Pepo234, 2020-04-12:
[image]

s0ftice, 2020-04-12:
if the modem's LED does not light up at all (not even flashing), you are probably just using a charging cable without data wires. You need a USB **OTG** cable.

Pepo234, 2020-04-12:
Si enciende pero siempre intermitente. Todas las veces que intento esta intermitente.

Pepo234, 2020-04-12:
> Si el LED del módem no se enciende (ni siquiera parpadea), probablemente solo esté utilizando un cable de carga sin cables de datos. Necesita un cable USB **OTG** .

If it turns on but always flashing.  Every time I try it is flashing.

ClassAxion, 2020-04-12:
Connect the modem to PC/laptop and check if it's working and connecting to the network properly.

Pepo234, 2020-04-12:
> Conecte el módem a la PC / computadora portátil y compruebe si funciona y si se conecta a la red correctamente.

Yes, connected is ok in the pc, don't have problems with pc but in disco I can't.

s0ftice, 2020-04-12:
telnet to your Disco and run `ulogcat |grep uavpal` after the modem has been plugged in for a minute or two. What is the output?

Pepo234, 2020-04-12:
> Telnet a su Disco y ejecutar `ulogcat |grep uavpal`después de que el módem se ha conectado durante un minuto o dos. ¿Cuál es el resultado?

Could you explain this to me?  When I do it?  At what point do I have to do it?

s0ftice, 2020-04-12:
Plug the modem into your Disco, turn it on, wait maybe 2 minutes, double press the Disco’s power button to enable telnet, telnet to your Disco, run the command, post the output here.

Pepo234, 2020-04-12:
[image]

s0ftice, 2020-04-12:
Have you installed the softmod?

s0ftice, 2020-04-12:
Please run the command `dmesg` what are the last 50 lines?

Pepo234, 2020-04-12:
> Ejecute el comando `dmesg`¿Cuáles son las últimas 50 líneas?

[image]

Pepo234, 2020-04-12:
> Ejecute el comando `dmesg`¿Cuáles son las últimas 50 líneas?

Pepo234, 2020-04-12:
> > Ejecute el comando `dmesg`¿Qué son son las últimas 50 líneas?

[image]

s0ftice, 2020-04-13:
please try a different cable

closed 2020-04-16 by s0ftice