uavpal/disco4g #74: Disco on Airmap.com

gelazanskas, 2020-04-18:
Hello I would like to request a feature - Parrot disco integration to Airmap.com.
It would be great to be able to see live telemetry (position on a map) from Disco on Airmap.com.
There is an API for that.
https://developers.airmap.com/docs/sending-telemetry

djoris, 2020-04-20:
A good start would be this:
https://github.com/uavpal/disco4g/blob/master/disco/uavpal/bin/uavpal_glympse.sh

Challenges:
1. Airmap's API is a bit difficult with the challenge-response, and buffered UDP. A flight ID should be sent. Perhaps a single Flight ID could be re-used by a drone, and thus simply stored on the drone?
1. Airmap's example code is not for sh, only php/python/java.
1. Unfortunately Airmap doesn't seem to allow transmitting messages or anything beyond the given paramters (position, speed..). That means no: battery status, 4G quality, ideally even a link to the video stream
1. the Glympse-script is't the prettiest to extract the telemetry variables from.

s0ftice, 2020-04-21:
@djoris 
[image]

Python on my Disco :)

closed 2020-05-17 by s0ftice