uavpal/disco4g #76: Link on install page to download softmod goes to old version (1.3)

rsilvers129, 2020-05-17:
https://github.com/uavpal/disco4g/wiki/Installation

Where it has a link to download the zip, it links to an old 1.3 version.

s0ftice, 2020-05-17:
the download links for 1.3 are in the link you already mentioned...
there is no new 1.3 or old 1.3. There is exactly one 1.3.

closed 2020-05-17 by s0ftice

rsilvers129, 2020-05-17:
The link says "Download .zip archive of the **latest** softmod scripts and binaries." 

Is 1.3 the latest version?

s0ftice, 2020-05-17:
yes it's the latest stable release.
https://github.com/uavpal/disco4g/releases

rsilvers129, 2020-05-17:
I see. Normally the Master branch is the latest stable release, and development work is done in other branches. This seems to be flipped.

"Why is the master so important to not mess with? One word: the master branch is deployable. It is your production code, ready to roll out into the world. The master branch is meant to be stable, and it is the social contract of open source software to never, ever push anything to master that is not tested, or that breaks the build. The entire reason GitHub works is that it is always safe to work from the master."

https://thenewstack.io/dont-mess-with-the-master-working-with-branches-in-git-and-github/

s0ftice, 2020-05-17:
The stable .zips are pulled from the releases, not from the master branch.

So even though master is v1.4-beta, people who install according to the instructions get 1.3-stable.

"never, ever push anything to master that is not tested, or that breaks the build"
1. who says what I push to the master if not tested?
2. in this case, nothing is built from the master branch, unless you deliberately decide to use it, see https://uavpal.com/disco/faq#beta

Regression can be done by creating a branch of a tagged release. This was never required so far.

The approach taken makes sense for smaller projects with little code contributors, it follows the KISS principle. :)
I do understand what you are saying, it's also mostly the approach I take. But for this project I deliberately decided against it.