uavpal/disco4g #81: disco4g

donato11320, 2020-08-07:
chi mi aiuta perchè non riesco a collegarmi con il disco parrot a zerotier non riesco a vedere nuovo utente o messo id
[image]
 di 16 cifre fatto tutto come nella guida la kiavetta si connette ed diventa blu lo Skycontroller 2 si connette a zerotier e lo vedo ma il disco no

djoris, 2020-08-09:
I don't speak Italian. Non parlo italiano.
Can you verify you have an internet connection from your Disco? Do you get a message on Pushbullet?
If you join https://uavpal.slack.com/ we might be able to help you faster.

donato11320, 2020-08-09:
because i can't connect with the parrot disk to zerotier i can't see new user or put id of 16 digits done everything as in the guide that is in the drescizione the kiavetta connects and turns blue the Skycontroller 2 connects to zerotier and I see but the disco does not want to know or loaded the softcam 5 times but nothing where am I wrong friend can you help me thanks

donato11320, 2020-08-09:
[image]

donato11320, 2020-08-09:
no I do not receive messages but the modem the parrot disk key is light blue if I put it in the pc it maybe connects and an error of the softmod that does not connect is making me lose my head the skycontroller 2 I see it on zerotier but the parrot disk does not he wants to know to connect

s0ftice, 2020-09-19:
did you get it to work?

OpenUAS, 2021-02-12:
As the issue description is also a bit on the undiscriptive side( "disco4g"), and no response for a while, maybe close the issue?( that is technically not even a software issue for this issue tracker.)

@donato11320 Hai ancora qualche ultima parola?

closed 2021-02-13 by s0ftice