uavpal/disco4g #82: Huawei E3372h-320 problem

ceufinger, 2020-09-20:
Hi,

I have the following Huawei modem:

Device name E3372h-320
Hardware version CL4E3372HM
Software version 10.0.3.1(H192SP21C00)
Web UI version WEBUI 10.0.3.1(W13SP12C03)
Configuration file version E3372h-320-CUST 10.0.3.2(C00)

Unfortunately, the Parrot Disco cannot connect to Zerotier and gives me the error message below in the log. Dooes somebody know what the issue could be? 

BusyBox v1.25.0 (2018-06-20 15:42:53 UTC) built-in shell (ash)                                                          Enter 'help' for a list of built-in commands.                                                                                                                                                                                                   / # ulogcat | grep uavpal                                                                                               I uavpal_drone(ulogger)                        : Huawei USB device detected (USB ID: 12d1:1f01)                         I uavpal_drone(ulogger)                        : === Loading uavpal softmod 1.3 ===                                     I uavpal_drone(ulogger)                        : ... detected Parrot Disco (platform evinrude), firmware version 1.7.1  I uavpal_drone(ulogger)                        : ... trying to use kernel modules compiled for firmware 1.7.0           I uavpal_drone(ulogger)                        : ... loading tunnel kernel module (for zerotier)                        I uavpal_drone(ulogger)                        : ... loading USB modem kernel modules                                   I uavpal_drone(ulogger)                        : ... loading iptables kernel modules (required for security)            I uavpal_drone(ulogger)                        : ... running usb_modeswitch to switch Huawei modem into huawei-new-mode I uavpal_drone(ulogger)                        : ... detecting Huawei modem type                                        I uavpal_drone(ulogger)                        : ... detected Huawei USB modem in Hi-Link mode                          I uavpal_drone(ulogger)                        : ... unloading Stick Mode kernel modules (not required for Hi-Link firmware)                                                                                                                    I uavpal_drone(ulogger)                        : ... connecting modem to Internet (Hi-Link)                             I uavpal_connect_hilink(ulogger)               : ... bringing up Hi-Link network interface                              I uavpal_connect_hilink(ulogger)               : ... requesting IP address from modem's DHCP server                     I uavpal_drone(ulogger)                        : Huawei USB device disconnected                                         I uavpal_drone(ulogger)                        : ... unloading scripts and daemons                                      E boxinit     (init)                           : untracked pid 1107 'uavpal_disco.sh' exited                            I uavpal_drone(ulogger)                        : ... clearing iptables rules                                            I uavpal_drone(ulogger)                        : ... clearing default route                                             I uavpal_drone(ulogger)                        : ... removing temp files                                                I uavpal_drone(ulogger)                        : ... removing lock files                                                I uavpal_drone(ulogger)                        : ... unloading kernel modules                                           I uavpal_drone(ulogger)                        : *** idle on Wi-Fi ***                                                  I uavpal_drone(ulogger)                        : Huawei USB device detected (USB ID: 12d1:1f01)                         I uavpal_drone(ulogger)                        : === Loading uavpal softmod 1.3 ===                                     I uavpal_drone(ulogger)                        : ... detected Parrot Disco (platform evinrude), firmware version 1.7.1  I uavpal_drone(ulogger)                        : ... trying to use kernel modules compiled for firmware 1.7.0           I uavpal_drone(ulogger)                        : ... loading tunnel kernel module (for zerotier)                        I uavpal_drone(ulogger)                        : ... loading USB modem kernel modules                                   I uavpal_drone(ulogger)                        : ... loading iptables kernel modules (required for security)            I uavpal_drone(ulogger)                        : ... running usb_modeswitch to switch Huawei modem into huawei-new-mode I uavpal_drone(ulogger)                        : ... detecting Huawei modem type                                        I uavpal_drone(ulogger)                        : ... detected Huawei USB modem in Hi-Link mode                          I uavpal_drone(ulogger)                        : ... unloading Stick Mode kernel modules (not required for Hi-Link firmware)                                                                                                                    I uavpal_drone(ulogger)                        : ... connecting modem to Internet (Hi-Link)                             I uavpal_connect_hilink(ulogger)               : ... bringing up Hi-Link network interface                              I uavpal_connect_hilink(ulogger)               : ... requesting IP address from modem's DHCP server                     I uavpal_connect_hilink(ulogger)               : ... setting eth1's IP address to                                       I uavpal_connect_hilink(ulogger)               : ... setting default route for .1                                       I uavpal_hilink_api(ulogger)                   : ... Error connecting to Hi-Link API                                    I uavpal_connect_hilink(ulogger)               : ... connecting to mobile network using APN "" (configured in the Hi-Link Web UI)                                                                                                               I uavpal_drone(ulogger)                        : ... enabling Hi-Link DMZ mode (1:1 NAT for better zerotier performance)I uavpal_hilink_api(ulogger)                   : ... Error connecting to Hi-Link API                                    I uavpal_drone(ulogger)                        : ... setting Hi-Link NAT type full cone (better zerotier performance)   I uavpal_hilink_api(ulogger)                   : ... Error connecting to Hi-Link API                                    I uavpal_drone(ulogger)                        : ... querying Huawei device details via Hi-Link API                     I uavpal_hilink_api(ulogger)                   : ... Error connecting to Hi-Link API                                    I uavpal_drone(ulogger)                        : ... model: , hardware version:                                         I uavpal_drone(ulogger)                        : ... software version: , WebUI version:                                 I uavpal_drone(ulogger)                        : ... applying iptables security rules for interface eth1                I uavpal_drone(ulogger)                        : ... starting connection keep-alive handler in background               I uavpal_drone(ulogger)                        : Huawei USB device disconnected                                         I uavpal_drone(ulogger)                        : ... unloading scripts and daemons                                      E boxinit     (init)                           : untracked pid 1363 'uavpal_disco.sh' exited                            E boxinit     (init)                           : untracked pid 1546 'uavpal_disco.sh' exited                            I uavpal_drone(ulogger)                        : ... clearing iptables rules                                            I uavpal_drone(ulogger)                        : ... clearing default route                                             I uavpal_drone(ulogger)                        : ... removing temp files                                                I uavpal_drone(ulogger)                        : ... removing lock files                                                I uavpal_drone(ulogger)                        : ... unloading kernel modules                                           I uavpal_drone(ulogger)                        : *** idle on Wi-Fi ***     [... cut ...] to Internet (Hi-Link)                             I uavpal_connect_hilink(ulogger)               : ... bringing up Hi-Link network interface                              I uavpal_connect_hilink(ulogger)               : ... requesting IP address from modem's DHCP server                     I uavpal_connect_hilink(ulogger)               : ... setting eth1's IP address to 192.168.8.119                         I uavpal_connect_hilink(ulogger)               : ... setting default route for 192.168.8.1

s0ftice, 2020-09-20:
this issue is known, the -320 model does not work with the softmod. try to get a -153 as per list of supported modems.

ceufinger, 2020-09-20:
Thanks for the quick answer!

ceufinger, 2020-09-20:
One more quick question: does the modem have to a version with the original Huawei software, or can it also be a version that is "branded" by a mobile network operator.

s0ftice, 2020-09-20:
it can also be branded. just make sure it's unlocked (not SIM locked to a particular operator)

tonysmartantzoulatos, 2021-05-28:
Have you tried installing the 153 FW to your 320 unit?

s0ftice, 2021-05-28:
have you? :)
the hardware is very different.

ClassAxion, 2021-06-01:
Probably installing -153 fw on -320 unit is impossible because updater checks the hw version.

MillenniusRex, 2021-06-09:
I am late to this party but am still interested in acquiring this tech. Is it possible to use this device in 2021? It’s my understanding that the US ban on Huawei services may make use of this impossible? If so, could there be an alternative?

s0ftice, 2021-06-09:
The US ban has no impact on Huawei modems. Your question has nothing to do with the topic of this issue. Can I ask you to join https://uavpal.com/slack instead?
Thanks

closed 2021-08-19 by s0ftice