uavpal/disco4g #83: Parrot Disco doesn't connect with ZeroTier

Jon290501, 2020-09-20:
Hello 
The Parrot Disco doesn't connect with ZeroTier after the Reboot on CMD. 
Modem --> E3372H-153 
There is NO Password on the Stick.
Installation on the Skycontroller 2 had work. 
The Skycontroller is visible on ZeroTier.

s0ftice, 2020-09-21:
can u send a photo of your modem without the front cover? (slide it aside)

Jon290501, 2020-09-21:
Hello 
The Problem was, that they send me the false version of the Huawei E3372....
Thank you for the quick answer and have a nice day ;) 

[image]

s0ftice, 2020-09-21:
Yeah, that's what I thought. I updated the docs today that the 320 model won't work. You are not the first one unfortunately.

closed 2020-09-21 by s0ftice