uavpal/disco4g #84: Geofence Option in FreeFlight Pro App missing

Marixlive, 2020-09-28:
Hello
Very nice work with this mod! But i have one question...
Do i need a older version of the FreeFlight Pro app? I can't find the option to disable Geofence under Setting/Safety.
I also searched all other settings without any luck.
I use the android version 5.2.7 of FreeFlight Pro.

Has anyone experience with this?

Cheers :)

Marixlive, 2020-10-11:
Hello
Okay sorry i need to reopen this issue. 
I tried to use the Free Flight 6 App but i aslo cant find the Geofence option there. Do you guys fly further than 2 km / 1.2 m?

Thanks for every advise. 
Cheers

s0ftice, 2020-10-11:
This is not really a Disco4G issue, I will close it therefore again.
See page 21 of the manual please.

[FreeFlight Pro User Guide.pdf](https://github.com/uavpal/disco4g/files/5361189/FreeFlight.Pro.User.Guide.pdf)

closed 2020-10-11 by s0ftice