uavpal/disco4g #85: Camera Only Connects When Using Wifi

jameslarson4thewin, 2020-11-07:
Hey ive completed all the steps for the mod, but seem to have an issue with getting the camera to connect when using 4g.  If i switch to wifi, the camera view shows up again in the freeflight app.  When i am in 4g, i get a grey screen.  Everything else appears to be working as it should with the mod.  If i push the button to take a snap shot or video, it does it, but i cannot see the pictures on the disco memory until i go back to wifi.  Then i can see the files.  If i try to see the files while in 4g mode, it says" Impossible to connect to device" with an red error message.
The light on the drone blinks blue, and the light on the controller is solid blue.  
Uploading some screenshots...
[image]
[image]
[image]
[image]

Any suggestions would be really great as to what i should do, because i have tried changing the camera settings a bit and nothing has fixed it so far...

Thank you.

s0ftice, 2020-11-09:
I believe you don't have sufficient bandwidth for the video stream. I suggest you do the two speed tests as advised here: https://uavpal.com/disco/faq#badconnection

jameslarson4thewin, 2020-11-09:
Thank you for the reply.  I do believe you are correct.  My Tmobile data is
awfully slow.  I am going to use two mint-mobile sim cards which i have
ordered.  This is all good news because its easily fixed.  And by the way,
you are a fucking legend for figuring this shit out and making these
tutorials which are so easy to follow.

s0ftice, 2020-11-11:
Thanks man :)

OpenUAS, 2021-02-12:
Close it? Since the FAQ already has advise to fix this ...kindly asked ;)

closed 2021-02-13 by s0ftice