uavpal/disco4g #87: Cant get USB MODEM to work.

yarnesbarnes, 2021-03-11:
Hi there,

Firstly congrats on amazing product, ive installed it over 50 times on different discos and have used it for years... My concern is I am fron Aus and I cant locate any compatiable usb modems ANYWHERE that work with softmod.. My Issue in detail is your list of huawei devices that are 'reported to work well with softmod' includes the Huawei 8372h, i have the telstra 8372h-320 & cant get it to work... any ideas??

geilde, 2021-03-11:
I have the  Huawei-E8372h-607  and it works.
Here is one for sale in Brisbane

https://www.ebay.com.au/itm/Huawei-E8372h-607-4G-LTE-4GX-WIFI-Mobile-Hotspot-Broadband-USB-Dongle-UNLOCKED/274686351964?hash=item3ff4951a5c:g:RXUAAOSwTfNgLGGs

s0ftice, 2021-03-12:
All the newer "320" models don't work (yet) with the softmod (i.e. E8372-320 and E3372-320)

This should be fine https://www.optus.com.au/shop/prepaid/mobile-broadband/usb/e3372

yarnesbarnes, 2021-03-12:
hey s0ftice i dont think that optus one will work, i tell you why because this optus one doesnt work ): 

Realise photos not the clearest but its a optus e3372h-608 which is same as optus prepaid you sent a link to ):

[image]
[image]
[image]

s0ftice, 2021-03-12:
this should work fine. what's the issue with it?
the Optus modem was recommended by one of our Aussie flyers. I suggest you join us on Slack: https://uavpal.com/slack the community is very helpful.

geilde, 2021-03-12:
What about what I suggested?

yarnesbarnes, 2021-03-12:
geilde 100% im getting one, im jsut so impatient... and regarding optus modem ill send you output when I run /data/ftp/uavpal/bin/uavpal_disco.sh  (thats file of memory that does connection)

yarnesbarnes, 2021-03-15:
hey i know this sounds crazy BUT could I edit the code to connect to a WiFi single as i do have a few wingle (wifi usb) sticks?? suggestions pls?

s0ftice, 2021-03-16:
as you mentioned yourself in the first comment, the E8372h is on the list of supported modems. This is a wingle.
Just the -320 version does not work. You would need to use an older model, like the E8372h-153, E8372h-607 or E8372h-608.

closed 2021-04-07 by s0ftice

s0ftice, 2021-07-10:
With regards to the E3372h-608: I think the -608 is a newer model just like the -320 that doesn't work. You would need to get your hands on a -607 or -153 model....